ProteoWizard/pwiz #1306: Vendor-format (and possibly other 3rd party) licenses

saifulbkhan, 2020-11-01:
Hello. I have been unable to find any documents or links where I can see the individual licenses under which the different vendors have allowed usage of their proprietary libraries. Is it not meant to be publicly visible?

My intention is to use pwiz for commercial purposes. I will not be modifying the source or redistributing it. More explicitly, I plan to use the docker (`chambm/pwiz-skyline-i-agree-to-the-vendor-licenses`), made available by the project itself, in a server setting. Is this permitted? A yes or no question here would suffice. This is my primary concern. If not, I am a little worried about the _"I agree to the vendor licenses"_ bit without actually going over the licenses.

Please let me know if this information is already available somewhere, and I have missed it.

chambm, 2020-11-02:
Short answer: yes.

Long answer: There are no restrictions about using ProteoWizard binaries in a commercial setting. The licenses are linked to at the download site for the regular installer (http://proteowizard.sourceforge.net/licenses.html), but you make a good point it should be on the Docker README as well.

closed 2020-11-02 by chambm

saifulbkhan, 2020-11-02:
Thanks @chambm!

chambm, 2020-11-02:
http://proteowizard.sourceforge.net/download.html See "I have read and agree to the license agreements".