ProteoWizard/pwiz #1310: msconvert - corrupted XML output?

fkromer, 2020-11-04:
I'm converting Agilent MassHunter raw data directories (directories ending with ".d") to mzML files. The msconvert command used for conversion is `msconvert <raw_data_dir_path> --64 --mz64 --inten64 -o <out_data_dir_path> --outfile <mzml_file_name>`. It popped up some issue `input ended before all started tags were ended; last tag started is 'chromatogramList'` when I tried to read (exclusively, no writes at all) the mzML files with pyOpenMS. This issue could be caused by corrupted mzML files generated with msconvert probably. However the issue does not necessarily have to relate to msconvert. The runtime environment beeing out of memory seems to be another possible root cause which leads to the pyOpenMS bad case behaviour observed. The raw data files are very small but I've to check the runtime utilization. Do you know about situations in which msconvert produces corrupted mzML files?

chambm, 2020-11-04:
It should be easy to narrow it down to pyOpenMS vs. msconvert. Look at the tail of the file. If it ends in `</mzML>` or `</indexedmzML>` then the file from msconvert is complete. Although if `<chromatogramList>` is the last start tag, it's quite possible that msconvert crashed when trying to convert the Agilent chromatograms. If that's the case then I'd need to see the .d data  to understand why it's failing.

fkromer, 2020-11-04:
@chambm Unfortunatelly the mzML file is not available anymore and the behaviour is no reproducible. Before spending a lot of time trying to reproduce the issue asking is first is probably better. In case you don't know about any situations files could be corrupted it's totally fine for me. Then I'll have to dive into investigation.

chambm, 2020-11-04:
OK you can reopen if you can repro or have more details.

closed 2020-11-04 by chambm

fkromer, 2020-11-04:
@chambm Thx a lot!