ProteoWizard/pwiz #1332: Python MSConvert script not converting all files

CaoimheFitzGibbon, 2020-11-26:
Hi,

I am using a script to convert my raw MS files to mzML format. It worked for all of my sample files and blanks but not for my QC pool or standards, even though all the files are in the same folder and I cannot understand the differences between them that would cause this issue. This is the script I have been given that has worked for other people in my lab group:

"
import os, sys
from subprocess import call
import glob

def MSConvertBatch():
    path = os.getcwd()
    
    for fn in os.listdir(path):
        if ".d" in fn:
    #for file in glob.glob("*.raw"):
            if os.path.isfile(path + '\\' + fn.replace('.d','.mzML')):
                print (path +"\\"+ fn + " already exists")
            else:
                print (path + '\\' + fn + " not converted yet!!") 
                os.system('"C:\Program Files\ProteoWizard\ProteoWizard 3.0.20321.6df943caa\MSConvert.exe"' + " " + fn +  " config.txt ")
                #print('"C:\Program Files\ProteoWizard\ProteoWizard 3.0.20321.6df943caa\MSConvert.exe"'+" " + fn + " config.txt ")
                print (path + '\\' + fn + " conversion completed!")
            
MSConvertBatch()
"

And this is the config.txt file contents:
"
mzML=true
zlib=true
filter="peakPicking true 1-"
filter="sortByScanTime"
"

When I run the script I see the progress of all the sample files being converted, but if I try using it on just the files it won't convert, the progress screen pops up for a second and disappears, without anything happening.

I would use the MSConvert app to convert them, but I read that the peakpicking algorithm would be different (with my script using a local maxima algorithm and the application using either vendor of CWT methods).

Any insights/help would be greatly appreciated, thank you.

chambm, 2020-11-26:
Your script is not using local maximum, it's using vendor centroiding ("true" means prefer vendor centroiding), which is pretty much the default.  I certainly recommend trying msconvert or the GUI outside of the python script on the non-converting files to try and get an error message for why it's not converting.

CaoimheFitzGibbon, 2020-11-27:
@chambm thank you very much, I was trying to read about the filter types from http://proteowizard.sourceforge.net/tools/filters.html but I must have misunderstood. My non-converting files converted fine (as far as I know) using the MSConvert GUI, and I used a modified config script to convert the rest. Do you happen to know if writing it as:

filter="peakPicking <vendor> <0.1> <0.1> 1-"

is the correct format for SNR and min peak spacing to be 0.1 and for it to be vendor centroiding? The files have been working for my XCMS script afterwards as well so I'm hoping it did work.

chambm, 2020-11-27:
To use CWT peak picking it would be:
`filter="peakPicking cwt snr=0.1 peakSpace=0.1`

closed 2021-02-17 by chambm