ProteoWizard/pwiz #1343: Wrong MS2 Spectrum annotation for Instrument: ACQ-SQD2#LCA870

sneumann, 2020-12-07:
Hi, 

I have data from an SQD2 (not sure what the LCA870 is)
https://www.waters.com/waters/en_US/SQ-Detector-2---Single-Quadrupole-Detection/nav.htm?cid=134631584

Function 1 is an ESI+ full scan, Function 2 is an API+ full scan. As you see below, 
the latter is annotated as an MS2 with CID=3eV from precursor "isolation" 800 +/- 700.

Unless the instrument has a secret TripleQuad upgrade, I don't trust the MS2 annotation. 

I checked `_extern.inf` and couldn't find any difference between funcs 1/2 causing 
this spurious MS2. 

Functions 3/4 are the same as ESI- and API-, and are all annotated correctly as MS1 in the mzML.

I'll provide the `*.raw` if I get the permission to do so. Or is there anything else 
I could check that would explain this behaviour ? 

Yours,
Steffen

```
        <spectrum index="3" id="function=1 process=0 scan=1" defaultArrayLength="1471">
          <cvParam cvRef="MS" accession="MS:1000579" name="MS1 spectrum" value=""/>
          <cvParam cvRef="MS" accession="MS:1000511" name="ms level" value="1"/>
          <cvParam cvRef="MS" accession="MS:1000130" name="positive scan" value=""/>
          <cvParam cvRef="MS" accession="MS:1000128" name="profile spectrum" value=""/>
          <cvParam cvRef="MS" accession="MS:1000504" name="base peak m/z" value="130.447342"/>
          <cvParam cvRef="MS" accession="MS:1000505" name="base peak intensity" value="8.7965696e07"/>
          <cvParam cvRef="MS" accession="MS:1000285" name="total ion current" value="6.414999552e09"/>
          <scanList count="1">
            <cvParam cvRef="MS" accession="MS:1000795" name="no combination" value=""/>
            <scan>
              <cvParam cvRef="MS" accession="MS:1000616" name="preset scan configuration" value="1"/>
              <cvParam cvRef="MS" accession="MS:1000016" name="scan start time" value="0.001700000023" unitCvRef="UO" unitAccession="UO:0000031" unitName="minute"/>
              <scanWindowList count="1">
                <scanWindow>
                  <cvParam cvRef="MS" accession="MS:1000501" name="scan window lower limit" value="100.0" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                  <cvParam cvRef="MS" accession="MS:1000500" name="scan window upper limit" value="1500.0" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                </scanWindow>
              </scanWindowList>
            </scan>
          </scanList>
```

```
        <spectrum index="6" id="function=2 process=0 scan=1" defaultArrayLength="797">
          <cvParam cvRef="MS" accession="MS:1000580" name="MSn spectrum" value=""/>
          <cvParam cvRef="MS" accession="MS:1000511" name="ms level" value="2"/>
          <cvParam cvRef="MS" accession="MS:1000130" name="positive scan" value=""/>
          <cvParam cvRef="MS" accession="MS:1000128" name="profile spectrum" value=""/>
          <cvParam cvRef="MS" accession="MS:1000504" name="base peak m/z" value="105.168823"/>
          <cvParam cvRef="MS" accession="MS:1000505" name="base peak intensity" value="3.846656e06"/>
          <cvParam cvRef="MS" accession="MS:1000285" name="total ion current" value="2.68558624e08"/>
          <scanList count="1">
            <cvParam cvRef="MS" accession="MS:1000795" name="no combination" value=""/>
            <scan>
              <cvParam cvRef="MS" accession="MS:1000616" name="preset scan configuration" value="2"/>
              <cvParam cvRef="MS" accession="MS:1000016" name="scan start time" value="0.003700000001" unitCvRef="UO" unitAccession="UO:0000031" unitName="minute"/>
              <scanWindowList count="1">
                <scanWindow>
                  <cvParam cvRef="MS" accession="MS:1000501" name="scan window lower limit" value="100.0" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                  <cvParam cvRef="MS" accession="MS:1000500" name="scan window upper limit" value="1500.0" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                </scanWindow>
              </scanWindowList>
            </scan>
          </scanList>
          <precursorList count="1">
            <precursor>
              <isolationWindow>
                <cvParam cvRef="MS" accession="MS:1000829" name="isolation window upper offset" value="700.0" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                <cvParam cvRef="MS" accession="MS:1000828" name="isolation window lower offset" value="700.0" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                <cvParam cvRef="MS" accession="MS:1000827" name="isolation window target m/z" value="800.0" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
              </isolationWindow>
              <selectedIonList count="1">
                <selectedIon>
                  <cvParam cvRef="MS" accession="MS:1000744" name="selected ion m/z" value="800.0" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                </selectedIon>
              </selectedIonList>
              <activation>
                <cvParam cvRef="MS" accession="MS:1000422" name="beam-type collision-induced dissociation" value=""/>
                <cvParam cvRef="MS" accession="MS:1000045" name="collision energy" value="3.0" unitCvRef="UO" unitAccession="UO:0000266" unitName="electronvolt"/>
              </activation>
            </precursor>
          </precursorList>
```

chambm, 2020-12-07:
It's interpreting it as MSe where the high energy function is treated as MS2. Waters doesn't provide a way to tell the actual instrument model or even the category (triple vs single quad vs qtof). Nor does it provide an easy way to differentiate between 2nd functions that are high-energy MSe from 2nd functions that are some other kind of MS1. Having some more data to look at would be helpful.

chambm, 2020-12-07:
Not to mention if the functions are coming from different ionization we should be annotating that in instrumentConfigurationRef.