ProteoWizard/pwiz #1381: Memory usage issues - DIA-Umpire pseudo-DDA spectra files

MarcIsak, 2021-01-04:
Hi, 

I have used the DIA-Umpire signal extraction filter in MSConvert to create pseudo-DDA-spectra successfully. I have though noticed that the memory usage is tremendous when using this filter. Creating a single pseudo-DDA spectra file (~500 MB, MGF format) from an input mzXML file (~ 1.3 GB) can take up > 14 GB of RAM on my machine (32 GB in total). I just wonder if it would be easy to enable for batch processing in order to reduce the memory footprint. Other conversions in MSConvert from RAW to mzML or mzXML does use much less memory for the conversion, where I typically convert 6 files simultaneously on my machine. 

I wish I could contribute more, but I have yet to learn how to code in C++. 

Best,

Marc

Best,

Marc

chambm, 2021-01-04:
Try the DIA-Umpire from this branch:
https://teamcity.labkey.org/repository/download/bt83/1212492:id/pwiz-setup-3.0.20365.00fd44114-x86_64.msi?guest=1

It uses spill files in your %TEMP% dir for each DIA window rather than keeping it all in memory. It's much more memory efficient that way.

MarcIsak, 2021-01-05:
Hi, 

thanks for the new update of the DIA-Umpire feature in MSConvert. I just tried it out on some files. It is indeed a little bit more memory efficient, as a single conversion from a mzXML file to pseudo-dda (MGF) file now takes up between 9-10 GB of RAM instead of 14-16 GB. Still, this time I could convert 2 files simultaneously on my machine which means I can convert all files in half the time. 

One thing I noticed was that the file sizes of the pseudo-dda files (MGF) are a little bit different as compared to the corresponding files I generated with the latest stable version of MSConvert (downloaded from ProteoWizard/SourceForge website). Are there other changes to the DIA-Umpire feature than the usage of spill files to reduce memory usage?

Best,

Marc

chambm, 2021-01-05:
I think the only difference is I added the quality level to the spectra as a userParam, which in the Java DIA-Umpire leads to each spectrum being written to one of 3 separate MGFs.

Our DIA-Umpire implementation is highly multithreaded and that works very well until the last few windows. Did you measure the overall conversion time when converting 2 files at once and found it to be half?

In the default configuration, the DIA windows are processed by #-of-cores/4 threads, and each thread can launch 4 threads to divide the DIA window's work up further. Each window builds its own list of peak curves, so memory usage is going to be a function of how many cores you have unless you limit the thread count. You can set the max # of threads like `Thread = 4` in the params file. There's also a `MultithreadOverWindows = false` parameter you could try which will only process windows one at a time, but will use as many cores as it can. Multithreading over windows was faster in my testing because it can parallelize when generating curves from the MS2 spectra (which have been completely loaded into memory). Running one window at a time (which is what the Java DIA-Umpire does), it can only parallelize the curve smoothing and clustering.

If there's a significant opportunity to conserve memory, it would probably be in avoiding loading the MS2 spectra into memory by generating the peak curves for all windows at once and writing them directly to files rather than keeping them in memory. But I'm skeptical that the memory conserved would offset the increased I/O, and I think a more urgent optimization is to figure out how to keep the last couple windows using all CPU cores.

MarcIsak, 2021-01-07:
Hi, 

I tried to set THREAD=4 in the DIA-Umpire parameters file, and then I ran 6 parallel(6 mzXML files) conversions successfully. The whole process now only takes 17 minutes. This is the best configuration I have found so far. Each conversion used less than 4 GB of RAM when lowering the number of threads to 4. So thank you for all the help so far 👍 .

Best,

Marc

closed 2021-01-15 by chambm