ProteoWizard/pwiz #1409: msConvert "Unknown Generic Error"

tomas-pluskal, 2021-01-14:
Hi, I am trying to convert a Waters Synapt G2 .RAW file which includes ion mobility separation into .mzML. The file comes from direct infusion, no chromatographic separation. msConvert fails with "Unknown generic error" - screenshot attached.

OS: Windows 7 Enterprise 64-bit
pwiz version: 3.0.21013.40426fb31
The Waters raw file is here: https://www.dropbox.com/s/aemntvzkilyri9z/20119%20AB2_IM.raw.zip?dl=0

[image]

chambm, 2021-01-15:
I think this should be solved by a newer Waters SDK; I'll ping my Waters contact to see if that's available yet.

chambm, 2021-01-21:
I just updated our Waters SDK but it doesn't solve this issue. The problem is your .raw file has an extra function in it which the _extern.inf file says shouldn't be there (there should only be function 1). Do you have any idea how it got there? If you move the FUNC002 files out of the .raw directory, pwiz can convert it fine. You're the 2nd or 3rd user to report an error caused by this, so I'm going to run it by my Waters contact to see if they can figure out what's causing it.

tomas-pluskal, 2021-01-21:
I got the file from our core facility, so I am not sure what is that extra function. But I guess the Waters people should know :)

chambm, 2021-02-01:
I still don't know why extern.inf doesn't include the extra function, but it seems we can actually read the data from it as long as we don't try to read drift time. We now the first function that returns a valid drift time for every function, because according to Waters all functions will have the same scan<->drift time mapping.

closed 2021-02-01 by chambm

tomas-pluskal, 2021-02-06:
Thanks for fixing it!

commit 49f5ef3d (2021-02-01): - fixed Waters "Unknown Generic Error" (#1429)

- fixed Waters "Unknown Generic Error" due to calling GetDriftTime() on functions existing in Waters .RAW but not listed in _extern.inf

Reported by Laura (Rozas?) and tomas-pluskal
Closes #1409