ProteoWizard/pwiz #1422: Have trouble when downloading ProteoWizard(Linux naive 64 bit) package from http://proteowizard.sourceforge.net/download.html

BBBBunnyZ, 2021-01-24:
Hey! I had trouble when downloading the ProteoWizard software package from http://proteowizard.sourceforge.net/download.html.
The version I need is Linux naive 64-bit. But when I clicked the "DOWNLOAD" button there is no response and I've tried this on many computers or laptops.
So I wonder if there is other way to download the Linux version of ProteoWizard?
Thank you soooooo much!

chambm, 2021-01-29:
Hi @BBBBunnyZ may I ask what network you're trying from?

corinasimian, 2022-11-29:
I am trying to download the software from the same URL. I am getting
```
<Error>
<Code>AllAccessDisabled</Code>
<Message>All access to this object has been disabled</Message>
<RequestId>YXSD5AZH7TFBFKSJ</RequestId>
<HostId>/57c/G8q3QSNndodn41q/iMxvlIbYYuPs+aC/C+y844Z0hVjr1Nw1yoYH7jBr1aFkRwdRRvJ+ZE=</HostId>
</Error>
```
for `https://proteowizard-teamcity-artifacts.s3.us-west-2.amazonaws.com/ProteoWizard/bt83/2138058/pwiz-setup-3.0.22331.4902f6a-x86_64.msi`

chambm, 2022-11-29:
Hi Corina,

We are sorting out some issues with our file host. Downloads will hopefully be fixed tomorrow. Thanks for reporting.

chambm, 2022-11-30:
Should be fixed.

closed 2022-11-30 by chambm