ProteoWizard/pwiz #1457: The --singleThreaded argument is not working for msconvert

alchemistmatt, 2021-02-16:
When calling msconvert.exe, our automation code uses `--singleThreaded`
Recent versions of msconvert report an error when that is used, specifically
```
Invalid command-line: option '--singleThreaded' requires at least one argument
```

If I instead change things and use `--singleThreaded=true` I see:
```
Invalid command-line: option '--singleThreaded' does not take any arguments
```

Am I calling things wrong or is this a bug?  This new behavior was likely introduced via pull request #1400 , commit 2b027525ae2781cfb601e13bb6b8ad698945e250

chambm, 2021-02-16:
Yes that's a bug. I'll see about fixing that. Good news is you probably don't have to use the flag anymore in your automation code since the default will usually be single threaded. Only CWT and turbocharger will use multiple threads.

alchemistmatt, 2021-02-16:
Yep, already updated:
https://github.com/PNNL-Comp-Mass-Spec/DMS-Analysis-Manager/commit/4b7e47cb4f18f6ecb85f40da0946ee48a03fac51

Thanks for confirming.

FarmGeek4Life, 2021-02-16:
Just looking and curious on this - is this because an "implicit_value" behavior isn't specified, when the implicit value behavior is probably built-in to the 'bool' handling?
https://www.boost.org/users/history/version_1_65_0.html
https://www.boost.org/doc/libs/1_67_0/doc/html/boost/program_options/typed_value.html#id-1_3_31_9_10_1_1_7_5_3-bb says:
typed_value * zero_tokens();
Specifies that no tokens may be provided as the value of this option, which means that only presense of the option is significant. For such option to be useful, either the 'validate' function should be specialized, or the 'implicit_value' method should be also used. In most cases, you can use the 'bool_switch' function instead of using this method.

chambm, 2021-02-17:
You may be right about `implicit_value` being needed. Without the default value it worked fine as a regular bool. Now it's a tribool (meaning there's no preference if the option is left unspecified), which means it can't really be a bool_switch or implicit_value. `--singleThreaded` or ` --singleThreaded=true` should both mean true, and `--singleThreaded=false` should mean false.