ProteoWizard/pwiz #1540: Error reading mzml files created by ThermoRawFileParser Version 1.3.3

hechth, 2021-04-19:
As already described [here](https://github.com/compomics/ThermoRawFileParser/issues/112), the new version of ThermoRawFileParser creates a CVTerm
```xml
<cvParam cvRef="MS" accession="MS:1003145" name="ThermoRawFileParser" />
```
 which is not recognized by Proteowizard and causes an error when reading the file.

edeutsch, 2021-04-20:
I also wonder if it would be good practice for msconvert not to halt if it encounters a term it does not recognize at minimum in the software/processing section, since by doing that, it is a disincentive for others to "do the right thing" and register a new tool in the CV and write out the fact that it was used in the production of an mzML (or potentially other format) file.

chambm, 2021-04-26:
I updated the CV last week so this file should work now. I agree it would be ideal if pwiz worked with newer CVs than it was compiled with but that would be quite tricky to implement. Easier would be to just monitor the CV and do an automatic update PR anytime there's an update. I just haven't gotten around to doing that. I am in the process of switching from Travis to GitHub actions though for some unofficial pwiz builds, so maybe I'll figure it out then.

chambm, 2021-04-28:
On further consideration, auto-updating pwiz when CVs change won't fix downstream tools that use pwiz, like mzR. Seems like runtime CV updating (when necessary) is the only way.