ProteoWizard/pwiz #1592: Unable to open custom mzML file

wkumler, 2021-05-13:
Hi all,
    I'm having some trouble using proteowizard tools with an mzML file that I've edited manually and am hoping you can give me some context for what I've done wrong. The error I'm getting is 

```
[SpectrumList_mzML::create()] Bad istream.
```

when running

```
msconvert --mzXML mini_mzML.mzML
```

I've attached the gzipped file to this issue so you can replicate it (~200kB). This file originally came from a Thermo Orbitrap, was converted to mzML by msconvert (successfully) and then edited to include a subset of the masses and intensities. I'm pretty sure the problem is that I don't know what to change the `defaultArrayLength` to after I've shortened the associated binary. I successfully edited the `binaryDataArray encodedLength` parameter to reflect the shorter binary, but can't figure out what the `defaultArrayLength` should be or if it needs to be changed. Everything else in the file is the same as the original.

I'm asking here because you seem to be the foremost mzML experts currently, and this is vaguely related to an earlier issue opened here (#1583). The [docs were unhelpful](http://www.peptideatlas.org/tmp/mzML1.1.0.html) and I'm not sure who else to ask. If this is too off-topic, please feel free to close.

[mini_mzML.mzML.gz](https://github.com/ProteoWizard/pwiz/files/6470094/mini_mzML.mzML.gz)

nickshulman, 2021-05-13:
If you want to hand edit a .mzML file, you should make sure to add the "--noindex" argument to the msconvert commandline when you create it.

The index appears at the end of the .mzML file, and specifies the exact byte offsets in the file where each spectrum can be found. It would be really difficult to fix all of those numbers when you edit the file, so it is much easier to edit a file which does not have an index.

By the way, I am not exactly sure what "defaultArrayLength" means, but my guess would be that it's the number of m/z and intensity values that are in the "binaryDataArrayList" element later on in the spectrum xml. You would not want to change the "defaultArrayLength" value without also changing the contents of the "<binary>" elements later on in the spectrum. Those <binary> tags contain base64 encoded zlib compressed numbers.

wkumler, 2021-05-13:
Brilliant! I'd forgotten about the index at the bottom of the file, and with your hint about the defaultArrayLength I was able to get it working. You were entirely correct - it's the number of m/z-int pairs in the scan. Thanks a bunch!

closed 2021-05-13 by wkumler

chambm, 2021-05-20:
You could use the threshold filter for this to pick the N most intense peaks.

wkumler, 2021-05-20:
@chambm Unfortunately, I'm looking for a few low-abundance compounds in a sample that has high levels of contaminants at other m/z masses. As I understand the threshold filter, setting a limit with that would select mostly my noisy contaminants and actually remove the low-abundance peaks that I'm interested in. Is there a way to remove known contaminants or specific m/z windows during file conversion with ProteoWizard? I've only seen a way to subset a single window with the mzWindow argument.