ProteoWizard/pwiz #1692: Centroid data off by 51ppm

eeko-kon, 2021-07-23:
Hello!

When I am using the CLI version with msconvert for thermo IDX data using the command line 
`msconvert *.raw --filter "peakPicking true [1 ,2]" --ignoreUnknownInstrumentError`
the peak picking seems to be off by quite a lot (51 ppm).. The GUI works fine, there's no error. Is there a new version I should use? Or is the command wrong for thermo data? Both MS1 and MS2 levels are retrieved in profile mode. 

Thank you!

Best,
Efi.

nickshulman, 2021-07-23:
Are you saying that you are getting different results when you use the command line program "msconvert.exe" versus when you use "msconvertgui.exe"? That would be very strange.

One thing that would be much less strange would be if you were getting different results between 32 bit and 64 bit ProteoWizard. If I remember correctly, the Thermo library that ProteoWizard uses for 32 bit is much older and works differently.

Can you send us one of your raw files?

eeko-kon, 2021-07-23:
https://drive.google.com/drive/folders/1_NF5hY5YI76jVxQnYpEtLGA7_JjbkYlX?usp=sharing

This is the raw file. Yes, this is exactly what I am saying, I am basically looking for an ion with exact mass 658.2126 and when I convert it to centroid data using the command line, this ion is now 658.227 or so. Which also generates other issues. Let me know what you think!

nickshulman, 2021-07-24:
Is there a particular spectrum in that raw file that we should look at?
Can you give me a screenshot of what your settings in MSConvertGUI looked like when you converted that file?
Also, tell me again what command line arguments you used with msconvert.exe

eeko-kon, 2021-08-17:
Hi Nick! Sorry for the delay, I just came back from summer holidays. You can find both the command line arguments and the ion of interest above. 

I am basically looking for an ion with exact mass 658.2126 and when I convert it to centroid data using the command line, this ion is now 658.227

Command line argument with MS2 centroid and MS1 profile data:

`msconvert *.raw --filter "peakPicking true [1]" --ignoreUnknownInstrumentError
`

eeko-kon, 2021-08-17:
[image]

Screenshot added

nickshulman, 2021-08-17:
Efi,

When I convert your file with the msconvert.exe commandline, I see the
warning message:
[SpectrumList_PeakPicker]: vendor centroiding requested but not available
for this data; falling back to local maximum peak picker

When I convert the file using msconvertGUI.exe, I do not see that message.

If msconvertgui was able to use vendor centroiding, and msconvert.exe was
not, that would explain the difference that you are seeing.
I have no idea why vendor centroiding would only be available in the GUI. I
will have to look into that, or maybe someone else on this list has an
answer.

By the way, is there a particular spectrum (retention time?) in the
converted mzML file that you would like us to look at? There are over 40000
spectra in that file, and I am not sure which spectrum you are looking at
when you see a difference regarding m/z 658.2126.

-- Nick

eeko-kon, 2021-08-17:
Ah, of course I understand your question. You will have to give me some time to go through the files again first thing in the morning :)

nickshulman, 2021-08-17:
Efi,

Your commandline arguments are not quite right. You are not supposed to put square brackets around the msLevel in the peakPicking filter.
It's supposed to be:

` --filter "peakPicking true 1"
`

If you put the square brackets around the number there, it gets interpreted as zero, so the warning I was seeing about vendor centroiding not being available was referring to the "electromagnetic radiation spectrum" spectra whose msLevel is 0.

Anyway, if you put square brackets around that number, your ms1 spectra will not be getting centroided. This might explain the difference that you are seeing between the gui and commandline.

-- Nick

chambm, 2021-08-17:
Nick is right, except it should be `--filter "peakPicking true 1-"` (ms level 1 or higher; otherwise it would just do ms level 1)

eeko-kon, 2021-08-17:
Thank you both! Good catch! I will run my files again in the morning and see if the error disappears. 
p.s. I was inspired from the instructions for CLI use of proteowizard for the command: 
https://proteowizard.sourceforge.io/tools/msconvert.html
```
msconvert data.RAW --zlib --filter "peakPicking true [2,3]"
(use zlib compression for binary data arrays, and use vendor centroiding for msLevels in [2,3])
```
I'm not sure if this is a mistake?

eeko-kon, 2021-08-18:
Alright, the issue is fixed, the masses are accurate :) Thank you so much! 

Just one more thing:

As I mentioned in another issue that I opened, I am constantly getting negative intensities in thermo converted files (both from GUI and CLI) which I am actually not getting when I use the **GNPS online converter.** What is the difference here? How can I avoid that error? 

p.s. 
1. My raw data do not have negative intensities
2. I used the correct CLI this time for the conversions and checked again

eeko-kon, 2021-08-18:
**GNPS converted_file.mzML**
File type: mzML
MS levels: 1, 2
Total number of peaks: 598907
Number of spectra: 3625

Ranges:
  retention time: 42.47 .. 660.22 sec (10.3 min)
  mass-to-charge: 50.00 .. 1961.75
  intensity:      61.06 .. 9333600256.00

Number of spectra per MS level:
  level 1: 605
  level 2: 3020

Peak type from metadata (or estimated from data)
  level 1: Centroid (Centroid)
  level 2: Centroid (Centroid)

Activation methods
    MS-Level 2 & HCID (High-energy collision-induced dissociation): 3020

Precursor charge distribution:
  charge 0: 1478x
  charge 1: 1222x
  charge 2: 319x
  charge 19: 1x

Number of chromatograms: 1
Number of chromatographic peaks: 3625

**CLI converted_file.mzML**
File type: mzML

Instrument: 

MS levels: 1, 2
Total number of peaks: 314649
Number of spectra: 41725

Ranges:
  retention time: 0.02 .. 762.00 sec (12.7 min)
  mass-to-charge: 0.00 .. 1961.84
  intensity:      -11350.93 .. 36974010368.00

Number of spectra per MS level:
  level 1: 38705
  level 2: 3020

Peak type from metadata (or estimated from data)
  level 1: Centroid (Centroid)
  level 2: Centroid (Centroid)

Activation methods
    MS-Level 2 & HCID (High-energy collision-induced dissociation): 3020

Precursor charge distribution:
  charge 0: 1478x
  charge 1: 1222x
  charge 2: 319x
  charge 19: 1x

Number of chromatograms: 2
Number of chromatographic peaks: 41725

I can clearly see that there is some filtering process in the GNPS conversion.. So it is normal to have negative intensities in the data and filter them out?

Thank you again. 
Efi.

eeko-kon, 2021-08-18:
I see! Great, thank you so much! 
Efi.

closed 2021-08-18 by eeko-kon