ProteoWizard/pwiz #1731: Not able to detect .baf file using MSConvertGUI

BenSamy2020, 2021-08-23:
Greetings,

Firstly I would like to thank you for providing an amazing open-source tool. I am trying to convert .BAF file to mzML. For some unknown reason, MSConvert is not able to detect the .baf files in my directory. I deleted my current version of MSConvert and re-downloaded the latest version. Despite using the latest version of MSConvertGUI, I am not able to detect the .baf file.

Please advise me on how to troubleshoot this issue. 

Regards,
Ben

nickshulman, 2021-08-23:
MSConvert needs all of the files in the ".d" directory, not just the .baf file.
When you press the "browse" button in MSConvertGUI, you are supposed to choose a folder whose name ends in ".d" and which contains a file named "analysis.baf". MSConvert uses the presence of the "analysis.baf" file to know that the folder is a Bruker data file folder, but the thing that you are choosing is the folder which contains the analysis.baf file.

-- Nick

BenSamy2020, 2021-08-23:
Greetings Nick,

Thank you for your rapid response. I am trying to download the data from PXD010092. Based on the FTP folder, I am not able to find any ".d" or "analysis.baf" file. Please refer to screen-shot for reference. I am sorry, this is my first time handling .baf files (I guess I might be doing a silly mistake...).

Regards,
Ben

[image]

chambm, 2021-08-23:
Yeah PRIDE messed that up. They should never have accepted only the BAF files. Known issue: https://github.com/PRIDE-Archive/px-submission-tool/issues/107 (same dataset, actually)

You can try to contact the authors to see if they have the original data.

closed 2021-08-23 by chambm

BenSamy2020, 2021-08-24:
Thank you @nickshulman and @chambm. Will contact the authors directly.