ProteoWizard/pwiz #1783: Precursor masses in mzML and mzXML files not correct

YasinEl, 2021-10-04:
I realized that when converting my Thermo .raw file to mzML or mzXML using Peak-Picking (and nothing else) via MSConvert GUI version 3.0.21277 (Windows) some precursor masses for MS2 scans are not correct (in the case I witnessed it was of by about 2 Da). I checked this using SeeMS, as well as MSnbase. However, the isotation Window is correctly displayed as in the .raw file. 
When checking in the raw file for the same scan the precursor mass correct there. In case this is important: I used an Inclusion list in ddMS2 acquisition mode on a QE-HF. 

I hope this can be helped. Thank you.

chambm, 2021-10-04:
Can you provide more detail about what the vendor software says and what the mzML says? 2 Da sounds like it could be the difference between monoisotopic peak and the most abundant isotope.

YasinEl, 2021-10-04:
In the Vendor software I get for scan Nr 2310: "FTMS + p ESI d Full ms2 98.0599@hcd45.00"
In SeeMS (mzML) I get for scan Nr 2310: "MS Level:2; Precursor info: 96.044261; Isolation Window: 97.559883-98.559883]

While it might be true that it is an isotope this does not explain why the precursor mass is not set to the actual precursor mass.

Thank you

chambm, 2021-10-04:
Hmm, the precursor m/z being outside the isolation window is a red flag. Can you share the RAW file?

But FYI the m/z in the filter line is usually NOT the same as what you'll see in SeeMS, which will be the "Monoisotopic M/Z" from the scan header, if it's available (present and non-zero). But it should always be in the isolation window.

YasinEl, 2021-10-04:
Ah I see. Then maybe this problem arises because I put specifically the mz value of the lower abundant isotope on the Inclusion list for the ddMS2 acquisition. I can share the raw file if you can give me an email or something to share it privately (Can I send it to some mail address?). (This is btw a significant problem more my workflow. Is there some way to still get the actual precursor mz value from mzML files?)

chambm, 2021-10-04:
OK I added my email to my GitHub profile. Hopefully I won't be inundated by spam. :)

YasinEl, 2021-10-04:
Ok I have sent you an email. 

Is there some other way to get the actual mz which has been fragmented from the mzML files instead of the monoisotopic mz?

chambm, 2021-10-04:
OK, so it's coding it properly. I don't quite understand why Thermo is saying the mono m/z is 96 when the window is 97.5-98.5, but I guess it's plausibly close?

The isolation target m/z, equal to the filter line m/z, is preserved in the mzML file. It's just not what SeeMS shows. If you're reading the mzML you should have no trouble seeing it.

```
<precursor spectrumRef="controllerType=0 controllerNumber=1 scan=705">
              <isolationWindow>
                **<cvParam cvRef="MS" accession="MS:1000827" name="isolation window target m/z" value="88.075607299805" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>**
                <cvParam cvRef="MS" accession="MS:1000828" name="isolation window lower offset" value="0.5" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                <cvParam cvRef="MS" accession="MS:1000829" name="isolation window upper offset" value="0.5" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                <userParam name="ms level" value="1"/>
              </isolationWindow>
              <selectedIonList count="1">
                <selectedIon>
                  <cvParam cvRef="MS" accession="MS:1000744" name="selected ion m/z" value="86.059933821369" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                  <cvParam cvRef="MS" accession="MS:1000041" name="charge state" value="1"/>
                  <cvParam cvRef="MS" accession="MS:1000042" name="peak intensity" value="3.356372820313e06" unitCvRef="MS" unitAccession="MS:1000131" unitName="number of detector counts"/>
                </selectedIon>
              </selectedIonList>
              <activation>
                <cvParam cvRef="MS" accession="MS:1000422" name="beam-type collision-induced dissociation" value=""/>
                <cvParam cvRef="MS" accession="MS:1000045" name="collision energy" value="45.0" unitCvRef="UO" unitAccession="UO:0000266" unitName="electronvolt"/>
              </activation>
            </precursor>
```

YasinEl, 2021-10-04:
Ah thank you. I never got the difference between that and the precursorMZ,.. So for the future: the precursorMZ is allowed to be outside the isolation window, but the isolationWindowTargetMZ (aka filter mz) is not,..

closed 2021-10-04 by YasinEl

speed090, 2025-09-10:
I have encountered the same issue. When converting Thermo .raw files to mzML, we found that some precursor masses for MS2 scans are not correct. I would like to know what precursor m/z and target m/z represent, and what the relationship between them is. Are they the same across all instruments?