ProteoWizard/pwiz #1793: Wiff2 conversion error

burn874, 2021-10-12:
Tried converting wiff2 using msconvert. Tried to fix it by adding the additional arguments which was suggested on other forums that may be causing issues however still no luck. Any idea on how to debug or what may be causing the issue?

`
docker run -it --rm \
        -v /nas/production/shared/admin/files/wiff2:/wiff2:rshared \
        pwiz/msconvert wine msconvert \
        --acceptZeroLengthSpectra \
        --ignoreMissingZeroSamples \
        --ignoreUnknownInstrumentError \
        /wiff2/lipid-210706.wiff2
`

> format: mzML 
>     m/z: Compression-None, 64-bit
>     intensity: Compression-None, 32-bit
>     rt: Compression-None, 64-bit
> ByteOrder_LittleEndian
>  indexed="true"
> outputPath: .
> extension: .mzML
> contactFilename:
> runIndexSet:
> 
> spectrum list filters:
> 
> chromatogram list filters:
> 
> filenames:
>   /wiff2\lipid-210706.wiff2
> 
> processing file: /wiff2\lipid-210706.wiff2
> [Reader_ABI::read()] unhandled exception
> Error processing file /wiff2\lipid-210706.wiff2

burn874, 2021-10-21:
I can convert it with the vendor software but not via pwiz/msconvert. Can't download the data due to regulations. Any other way for me to debug or obtain logs for you?

chambm, 2022-01-11:
Is it an XTOF? If not the SDK they gave us doesn't currently support it, but IIUC in that case there should also be a WIFF1 file for the same acquisition.

burn874, 2022-01-11:
I have no idea, do you want to take a look at it instead? I have absolutely no experience in this field.

chambm, 2022-01-11:
You can contact the people you got the data from to ask what instrument it comes from (that should be a very easy question to answer). If it's not from X500 QTOF then our WIFF2 SDK can't open it, so in that case having the file wouldn't do much good. But if it is from X500 then I'll be happy to take a look. If it's not from an X500 then you should ask the people you got the data from for the WIFF1 version of the data.

burn874, 2022-01-11:
7500 Qtrap. It's not supported right? Do you have any idea on how to extract this?

burn874, 2022-01-11:
Or what's the difference between those two. Is it possible to export it as the x500 version?

chambm, 2022-01-11:
The bad news is we don't support WIFF2 from the instrument. The good news is we're expecting an API update to support the 7500 and 7600 in the next month or so. The (hopefully) better news is you should be able to get a WIFF1 version of that acquisition from your user, and msconvert supports WIFF1 from any instrument AFAIK.

burn874, 2022-01-11:
Alright thanks. I'll close the issue.