ProteoWizard/pwiz #1872: Is there a way to get Absorbance data from .WIFF file from SCIEX QQQ 6500 Instrument

weiran-wang, 2021-12-02:
Hi, thank you for this amazing tools. I'm working on converting some LCMS data from a QTRAP 6500 by using the latest MSConvert. But is there a way to include absorbance data in the exported TXT (or other clear txt format) files? 
Thank you!

chambm, 2021-12-02:
Can you share the WIFF file? I don't think I've ever seen UV data in a WIFF so I don't know how to pick it out programmatically. We already support pressure and flow rate chromatograms from WIFF.

weiran-wang, 2021-12-02:
I can definitely share one wiff and wiffscan file here, is there a good way to share these files with you?

weiran-wang, 2021-12-02:
Besides we can definitely view the absorbance data in Analyst and SciOS.

chambm, 2021-12-02:
How big are the files? Send me an email (view my profile) and I'll send you a link to upload to.

weiran-wang, 2021-12-02:
Thank you Matt, it's about 15 MB

chambm, 2021-12-02:
You can zip them and attach to this issue.

weiran-wang, 2021-12-02:
[Mass Data Wiff.zip](https://github.com/ProteoWizard/pwiz/files/7644153/Mass.Data.Wiff.zip)

weiran-wang, 2021-12-02:
I chose a smaller one, and this should contains the absorbance data.

chambm, 2021-12-02:
It seems there are 2 operating modes: signal and spectrum. The file you attached is in spectrum mode. With no specific wavelengths to target, it seems I can only put a TWC, or output all the EMR spectra? Do you have a signal mode file?

weiran-wang, 2021-12-02:
No we don't have a signal mode data. And I don't think there is no such mode in the instrument. TWC should stands for Total Wavelength Chromatogram, I think this is what I saw from Analyst.

chambm, 2021-12-02:
Is that useful for analysis, or just QC? I would think you would want to look at specific wavelengths for doing any analysis. (I know almost nothing about UV-based analysis)

weiran-wang, 2021-12-02:
I would say yes, UV or TWC also tells you you have something flow through the system at that time. In our case we just want to see if we are missing anything in this sample. But some time, say you are not super sure about what's in this sample, you definitely wants to check the absorbance data. And sometimes you will see: hey I have a absorbance peak here at 2 min, but I cannot see that peak from my TIC/MS figure, what could happen?

weiran-wang, 2021-12-03:
To add that Absorbance and or TWC should be a very important data collected from a LCMS run.

chambm, 2021-12-03:
Thanks, I will add TWC when I get a chance and then close this issue.

PR #1889 (2021-12-09): - added total wavelength chromatogram for WIFF files with UV (DAD) detector
Closes #1872
closed 2021-12-09 by chambm