ProteoWizard/pwiz #1888: MS2 not being reported with Waters MSe conversion from command line

rhballachay, 2021-12-08:
Hello, I am using a Water's Synapt G2-Si with MSe acquisition and msconvert from the command line. I have discovered that the mzml file does not contain any scan level information for MSe. Aka, when using the command line, all spectra (whether MS1 or MSe, are all called MS1). I confirmed this using the data from [MSe_Short.raw](https://github.com/ProteoWizard/pwiz/tree/master/pwiz/data/vendor_readers/Waters/Reader_Waters_Test.data/MSe_Short.raw). The following was present only when using the GUI: <cvParam cvRef="MS" accession="MS:1000827" name="isolation window target m/z" value="1025.0" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>. I have also attached the sample files (converted to txt) for reference.Having this functionality in command line would be greatly appreciated. 

[MSe_Short_GUI.txt](https://github.com/ProteoWizard/pwiz/files/7679069/MSe_Short_GUI.txt)
[MSe_Short.txt](https://github.com/ProteoWizard/pwiz/files/7679071/MSe_Short.txt)

nickshulman, 2021-12-08:
I think the problem might just be that you have a very old version of the commandline msconvert.exe.

It looks like MSe_Short_GUI.txt was created with ProteoWizard 3.0.21342 and MSe_Short.txt was created with ProteoWizard 3.0.11242.
Version 3.0.11242 is from 2017.

When I convert that file using a more recent version of msconvert (3.0.21259) I correctly see that spectrum as being MS2.

rhballachay, 2021-12-09:
Thanks, that resolved it. Guess the problem was caused by OpenMS, not Proteowizard directly. Seems that OpenMS uses (old) proteowizard binaries and adds them to system variables on download. Even when you add the new proteowizard binaries to PATH afterwards, the OpenMS binaries are used by default.

closed 2021-12-09 by rhballachay

chambm, 2021-12-09:
The PATH is searched from left to right so if the OpenMS one comes before the newer one you added, it would be used instead of the newer one. You can prepend like:
`set PATH=c:\path\to\binaries;%PATH%`