ProteoWizard/pwiz #1905: [New Feature] Option to force rewrite of nativeid to be sequential scan numbers in mzML and mzXML

mwang87, 2021-12-14:
This is generally a really big issue within Agilent and Sciex data. Likely should be off by default but would be super helpful for downstream data analysis.

chambm, 2021-12-14:
The whole idea of the nativeID is that it points uniquely to a vendor spectrum the way their own software and/or API does. For Waters and Sciex that simply cannot be a single number. Downstream software that wants to work with nativeIDs must store them as strings (or parse them into multi-integer structs). Or it can just read the spectrum indices. You can convert to mzXML or MGF if you want to force pwiz to lose the original nativeID and effectively fall back to index+1 as the scan numbers (for mzXML).

closed 2021-12-14 by chambm

mwang87, 2021-12-14:
Thanks Matt!

I understand the principle underlying the nativeIDs, we'll just make downstream processing that forces the renumbering especially because in the practical sense, Agilent identifiers never work to map back to the raw data anyway.

mwang87, 2021-12-14:
Yeah, I think I'm in that latter camp of working outside their specific API. I'll be chatting with their software team later this week, so I'll bring up the issue and see if I can see if there are some alternatives that can make it more smooth.

chambm, 2021-12-14:
Scan start time is often unique as well and can be used as a lookup in some APIs (forgot if MHDAC is one of those). If you're not using their API, what are you using?

mwang87, 2021-12-14:
A few collaborators were looking at data in Agilent software via the GUI. We aren't accessing data programmatically.

chambm, 2021-12-14:
OK, scan time should work in that case, right?

mwang87, 2021-12-14:
I mean thats not really the goal though. We generally don't care to reference it back to the original vendor file, so concordance to the nativeID doesn't really matter. We are looking forward and making it easy for existing tools to interact with the mzML file, and sequential scan numbers would go a long way to helping with that.