ProteoWizard/pwiz #1943: Issues recognizing ms levels in mzML files

mwang87, 2022-01-12:
When converting and trying to filter for MS 1 and MS2 spectra, msconvert doesnt seem to recognize the scans as being so and returns an mzML that is empty. 

The file is here: 

https://massive.ucsd.edu/ProteoSAFe/DownloadResultFile?file=f.MSV000087434/ccms_peak/4-Liver-targeted-MS2-mzxml/4G-20200327-msms-std/0327-pos-ms2-std-211.mzML&forceDownload=true

and here is the command using the latest version of proteowizard:

```
./msconvert 0327-pos-ms2-std-211.mzML--mzML --32 --outfile temp/8d99d0a1-a61c-449e-b23d-e91fb7576b9b.mzML --outdir temp --filter 'threshold absolute 1 most-intense' --filter 'msLevel 1-4'
```

This is puzzling because in the input mzML, it clearly specifies the mzML ms level:

```
<spectrum index="1" id="controllerType=0 controllerNumber=1 scan=2" defaultArrayLength="4" dataProcessingRef="pwiz_Reader_conversion">
          <cvParam cvRef="MS" accession="MS:1000511" name="ms level" value="2"/>
```

chambm, 2022-01-12:
Hi Ming,

That RAW file apparently went through a buggy mzXML step before it got to mzXML (i.e. RAW -> mzXML -> mzML). That's not ideal even if the mzXML is perfect, but in this case the converter used to write the mzXML wrote the wrong `scanType` attribute. If it was missing entirely ProteoWizard would assume MSn, but if it's a bad value it just ignores that value and doesn't write any spectrum type at all. That makes for an invalid mzML file and as you discovered the filters might not work as expected on them (in this case the ms level filter thinks it's not an MS spectrum at all). Nevertheless I can easily add a workaround for this so that bad values also assume full scan but issue a warning.

mwang87, 2022-01-12:
Thanks Matt for the explaination and that makes a lot of sense. Looking at the mzXML seems like the original author uploaders must have converted with some other piece of software. 

I also agree I think issuing a warning and assuming full scan is likely the way to go.

Thanks!

Ming