ProteoWizard/pwiz #1946: With .d as input, the latest msconverter does not write mean inverse reduced ion mobility array array to the mzML file.

fcyu, 2022-01-15:
With the old msconverter (e.g. 3.0.21016), there are `mean inverse reduced ion mobility array` in the mzML converted from .d ddaPASEF. However, with the latest version, that array is gone, which crashes the MS1 quantification in IonQuant (https://github.com/Nesvilab/FragPipe/issues/581).

Could you please tell me why removing ion mobility arrays from the mzML file? It is needed in the MS1-based quantification.

Thanks,

Fengchao

chambm, 2022-01-16:
You converted with combineIonMobility on? The extra array should still be there. Note that scanSumming, often used with ddaPASEF, will drop extra arrays because it doesn't know how to sum the extra arrays and in the case of ion mobility it doesn't make sense to sum or average the ion mobility values anyway.

fcyu, 2022-01-16:
Thanks for your prompt reply. It explains everything.

May I give a small suggestion? Can you add the scanSumming filter to the `dataProcessingList`? Currently, it is not there with scanSumming enabled:

```xml
    <dataProcessingList count="1">
      <dataProcessing id="pwiz_Reader_Bruker_conversion">
        <processingMethod order="0" softwareRef="pwiz_Reader_Bruker">
          <cvParam cvRef="MS" accession="MS:1000544" name="Conversion to mzML" value=""/>
        </processingMethod>
        <processingMethod order="1" softwareRef="pwiz_Reader_Bruker">
          <cvParam cvRef="MS" accession="MS:1000035" name="peak picking" value=""/>
          <userParam name="Bruker/Agilent/CompassXtract peak picking"/>
        </processingMethod>
        <processingMethod order="2" softwareRef="pwiz_Reader_Bruker">
          <userParam name="removed extra zero samples"/>
        </processingMethod>
      </dataProcessing>
    </dataProcessingList>
```

Thanks,

Fengchan