ProteoWizard/pwiz #1954: mzXML parsing doesn't support mzML source files

rickhelmus, 2022-01-20:
Hi guys,

The `translateSourceFileTypeToNativeIdFormat()` function doesn't translate the `MS_mzML_format` flag ([source](https://github.com/ProteoWizard/pwiz/blob/master/pwiz/data/msdata/Serializer_mzXML.cpp#L872)). This may cause errors when trying to load `mzXML`  files that were generated from `mzML` files by some tools (e.g. `FileConverter` from OpenMS).

A workaround is to 'convert' the output `mzXML` to another `mzXML`, which changes the parent format to `mzXML`, which can then be parsed by `translateSourceFileTypeToNativeIdFormat()`.

chambm, 2022-01-20:
Hmm, there's no clearly correct approach here. The goal for this function is to recover vendor-specific nativeIDs from mzXML's single scan numbers to make it somewhat round-trip-able. But the nativeID format for an mzML file cannot be determined without looking inside the mzML (or having that extra information encoded in the mzXML). So I've committed a cop-out fix.

rickhelmus, 2022-01-21:
I see ... thanks for the details and the fix!

Out of curiosity: I am not fully sure what it means to lack the `nativeID` information here, and what this means for the possibility to 'safely' convert between mz(X)ML files.

Thanks,
Rick

chambm, 2022-01-21:
Some vendors have spectrum ids that can be practically treated as single numbers, and some do not. For the ones that do, ProteoWizard tries to interpret mzXML scan numbers as those ids. For example, Thermo has a nativeID format like `controllerType=0 controllerNumber=1 scan=123` because PDA spectra can have the same scan numbers as MS spectra but will be on a different controllerType. But because it's rare to find a RAW file with PDA spectra and mzXML doesn't support PDA spectra anyway, we can assume that mzXML files with a Thermo RAW parentFile should have their scan numbers translated back into Thermo nativeIDs. But if the conversion chain loses track of the fact the original vendor file was a Thermo RAW file, we can't do that translation.

rickhelmus, 2022-01-24:
Hi Matt,

That makes sense, many thanks for your explanation.

PR #1955 (2022-01-24): * fixed reading mzXML from mzML parentFile
There's no way to know the correct nativeID type so just fall back to scan number only.

Closes #1954
closed 2022-01-24 by rickhelmus