ProteoWizard/pwiz #2026: Build on Ubuntu and use msconvert

maxvincent24, 2022-03-09:
Hi all,

I use a docker image based on Ubuntu 20.04 for datascience (with Python, R, Julia, JupyterLab, slurm) and would like to install Proteowizard to convert .d files to .mzML with msconvert. I downloaded the source, bjam build which includes vendor reader support, then put the unzipped pwiz-src-3_0_22067_6471540 folder inside the mounted volume of my docker image. From inside the image, I ran `./quickbuild.sh` to build Proteowizard.

Once the build is done (after more than an hour), I don't understand what to do for the 6th point of the instructions [link](https://proteowizard.sourceforge.io/user_installation.shtml):
`6) For convenience, adjust your "PATH" environment variable to include your ProteoWizard tools directory, so that the command line tools may be run from any directory.`

I also don't get how to use msconvert once the build is done :(

PS :
I am not even sure the build went well, I got the following message at the end :

```
...failed updating 3 targets...
...skipped 270 targets...
...updated 1601 targets...
At least one pwiz target failed to build.
```

And during the build, I got many lines starting with `...skipped `, especially the following ones about msconvert :
```
...skipped <p/home/jovyan/work/Proteowizard_source/pwiz-src-3_0_22067_6471540/build-linux-x86_64/pwiz_tools/commandline/gcc-9/release/link-static/runtime-link-static/threading-multi>msconvert for lack of <p/home/jovyan/work/Proteowizard_source/pwiz-src-3_0_22067_6471540/build-linux-x86_64/pwiz/data/identdata/gcc-9/release/link-static/runtime-link-static/threading-multi>libpwiz_data_identdata.a...
...skipped <p/home/jovyan/work/Proteowizard_source/pwiz-src-3_0_22067_6471540/build-linux-x86_64/gcc-release>msconvert for lack of <p/home/jovyan/work/Proteowizard_source/pwiz-src-3_0_22067_6471540/build-linux-x86_64/pwiz_tools/commandline/gcc-9/release/link-static/runtime-link-static/threading-multi>msconvert...
gcc.link /home/jovyan/work/Proteowizard_source/pwiz-src-3_0_22067_6471540/build-linux-x86_64/pwiz_tools/sld/gcc-9/release/link-static/runtime-link-static/threading-multi/short_dump
gcc.compile.c++ /home/jovyan/work/Proteowizard_source/pwiz-src-3_0_22067_6471540/build-linux-x86_64/pwiz_tools/commandline/gcc-9/release/link-static/runtime-link-static/threading-multi/msdir.o
```

I also spotted these ones :
```
gcc.compile.c++ /home/jovyan/work/Proteowizard_source/pwiz-src-3_0_22067_6471540/build-linux-x86_64/pwiz/data/identdata/gcc-9/release/link-static/runtime-link-static/threading-multi/IdentData.o
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
```

Would there be a way to install/build only msconvert and not the entire Proteowizard tool ?

Thanks so much in advance for your help !
Maxime

chambm, 2022-03-09:
Hi Maxime,

The build did not succeed, but that doesn't matter because you want vendor support and native Linux builds don't provide that. You need to include Wine in your docker image and like we do with the official ProteoWizard Docker container:
https://github.com/ProteoWizard/container/blob/master/Dockerfile

Unless you want to rebuild your container to use Ubuntu 16.04 (the official update to 20.04 got reverted because it wasn't passing all tests), I think you best bet may be to have your docker container call our official Docker container when you want to run msconvert. Whether you do that with Docker-in-Docker or some kind of compose set up is up to you.

maxvincent24, 2022-03-28:
Hi Matt,

Thanks you for your helpful answer. After a few days to find a solution, I decided my initial idea was not the best way to use your container so I'll go with an other way !

Thanks again !

chambm, 2022-03-28:
I'm glad you found a way to solve your problem!

closed 2022-03-28 by chambm