ProteoWizard/pwiz #2028: Compile error on clang 11,12 and later

sneumann, 2022-03-14:
Hi, 
we had reports of compile issues on `clang` https://github.com/sneumann/mzR/issues/244 which resurfaced after upgrading to a new pwiz codebase https://github.com/sneumann/mzR/issues/263, where `clang` complains about 
https://github.com/ProteoWizard/pwiz/blob/6ea7fb3eeeaecc37d80aa3703afbcdabcbda3e56/pwiz/data/common/BinaryIndexStream.cpp#L157
with the message
```
clang++ -std=gnu++11 -I"/data/gannet/ripley/R/R-clang/include" -DNDEBUG -I./boost_aux/ -I./boost/ -I. -D_LARGEFILE_SOURCE -DHAVE_PWIZ_MZML_LIB -D_NODEBUG    -I'/data/gannet/ripley/R/test-clang/Rcpp/include' -I'/data/gannet/ripley/R/test-clang/Rhdf5lib/include' -isystem /usr/local/clang/include   -fpic  -g -O3 -Wall -pedantic -frtti  -c pwiz/data/common/BinaryIndexStream.cpp -o pwiz/data/common/BinaryIndexStream.o
pwiz/data/common/BinaryIndexStream.cpp:159:15: error: cannot assign to non-static data member within const member function 'operator[]'
        next_ = _Off * value_size_;
        ~~~~~ ^
```
Back then, Kurt Hornik suggested to remove the `const`, which solved the issue in our case. 
As I am not a C++ wizard, I can't tell if there is a more correct way to fix the compiler error by changing how `next_` is declared in 
https://github.com/ProteoWizard/pwiz/blob/6ea7fb3eeeaecc37d80aa3703afbcdabcbda3e56/pwiz/data/common/BinaryIndexStream.cpp#L206

Yours, Steffen

chambm, 2022-03-14:
I'm not even sure what that member function is there for. It doesn't seem to be used anywhere in the code. Making it non-const doesn't seem right, so just removing it outright is probably what I'll do.

chambm, 2022-03-14:
I think the reason no other compiler have an error for it is because it's not used. Later Clangs are apparently pretty proactive about checking unused code!

closed 2023-02-09 by chambm