ProteoWizard/pwiz #2031: Compile independent of bundled dependencies

bernt-matthias, 2022-03-15:
Is there a possibility to get ProteoWizard compiled without the bundled dependencies (boost, etc). 

This would be extremely handy for things like bioconda https://github.com/bioconda/bioconda-recipes/blob/master/recipes/proteowizard/meta.yaml where currently only the binaries are installed (also headers and libraries should be available there). 

This would allow the use in other tools like: https://github.com/statisticalbiotechnology/maracluster

I understand that this would come without vendor support, but this would be OK, since users should convert vendor formats to standard formats as a first step of the analysis anyway.

chambm, 2022-03-15:
It's not clear to me what you're asking for. Our official builds statically link to most of their non-vendor dependencies so even if we were to provide (.SO) libraries in the bioconda recipe (which I should think would be a different recipe entirely), the binaries would still have those dependencies "bundled". If you're talking about .A libraries, then yes it's possible to extract those out. I don't think the `libraries` target (e.g. `quickbuild.sh libraries`) is working right now though. I'll look into it.

I don't think ProteoWizard is going to compile with Boost <= 1.60 BTW.

bernt-matthias, 2022-03-15:
The question is not so much about static / dynamic linking (might be another question :) .. but maybe the questions are connected ). 

The potential problem is the following: lets say that ProteoWizard is linked statically to boost version x. Now there is other software that uses libraries of ProteoWizard and also boost version y (provided by conda or bundled). Then it seems at least nasty to mix boost versions in this way. 

> If you're talking about .A libraries, then yes it's possible to extract those out. I don't think the libraries target (e.g. quickbuild.sh libraries) is working right now though. I'll look into it.

This would be cool .. maybe that is a good first step to have ProteoWizard install libraries and headers.

chambm, 2022-03-15:
Yes that is a pretty common problem (source of ODR violations) with statically linking Boost and other ubiquitous libraries. There's actually a tool, bcp, that can rename all the boost namespaces to something else (e.g. boost_pwiz) so that multiple versions can coexist without violating ODR. Main problem being you have to redo that every time you update Boost. Admittedly, we update Boost very infrequently so it wouldn't be a big burden if there was a script to apply it. PRs welcome ;) Dynamic linking is an alternative solution but has its own issues. I've been looking at C++ modules (C++20 feature) but it's not clear to me that they would solve this issue either.