ProteoWizard/pwiz #2087: How to keep the value of collision energy

MuyaoXi9271, 2022-04-29:
Hi expert,

I would like to consult how to set parameters to keep the values of collision energy of MS2 DDA data after transferring into the format of .mzml. I have tried many settings, ending up with zero for collision energy. I appreciate any help very much.

Bests,
Muyao

chambm, 2022-04-29:
By MS2 DDA data do you mean the .ms2 format?

MuyaoXi9271, 2022-05-01:
> By MS2 DDA data do you mean the .ms2 format?

Not really. The format of the raw file is .d and I would like to transfer raw data into the format of .mzml or .mgf. The issue is that I am not able to keep the value of collision energy by adjusting the parameters after transformation.

Bests,
Muyao

chambm, 2022-05-02:
There are at least 5 ".d" formats: Bruker BAF, Bruker TDF, Agilent/Bruker YEP, Agilent MassHunter, and Agilent Chemstation. Each one has a different API which may or may not support extracting collision energy. Which format is your data?

MuyaoXi9271, 2022-05-02:
The file is exported from Bruker tims-tof and I check the subfolder where there is a file called analysis.baf. Therefore, I guess it should be Bruker BAF.

Does msconvert support extracting collision energy in terms of this type of file?

Bests,
Muyao

chambm, 2022-05-02:
TimsTOF should be TDF format. You should have an analysis.tdf file in there. I'm not sure what the analysis.baf would be for. You should be getting collision energy for TDF format. So this is DDA PASEF data? If you're not getting collision energy in the precursors I'd like to see the analysis.tdf file.

MuyaoXi9271, 2022-05-02:
I am not so familiar with whether it is the DDA PASEF data. Could you please guide me on how to check whether it is the DDA PASEF data? The below image shows the contents of the subfolder of the .d file
[image]

chambm, 2022-05-02:
Can you zip up and attach the .sqlite file to this post? I'm pretty doubtful that is really from a timsTOF.

MuyaoXi9271, 2022-05-03:
Sure, I will send it to you personally, regarding GNPS, I would not like to share it in public.

MuyaoXi9271, 2022-05-03:
Sure, I will send it to you personally, regarding GDPR, I would not like to share it in public.

MuyaoXi9271, 2022-05-04:
> Can you zip up and attach the .sqlite file to this post? I'm pretty doubtful that is really from a timsTOF.

Hi Matt Chambers,

Did you receive the zip file?

Bests,
Muyao

MuyaoXi9271, 2022-05-04:
Thanks very much. I am looking forward to hearing good news.

Bests,
Muyao

closed 2022-05-06 by chambm

MuyaoXi9271, 2022-08-24:
Hi Matt Chambers,

It works and the value of collision energy is available for the positive mode, but they are not available for the negative mode. Do I need to provide some files for you to fix it?

Bests,
Muyao

PR #2095 (2022-05-06): Get CE from Bruker BAF format
- increased maximum worker thread count for msconvert to 16
- added support for getting CE from Bruker BAF format

Closes #2087