ProteoWizard/pwiz #2121: Msconvert failed with Agilent Mass Hunter (.d) data

jjacobson95, 2022-06-03:
Hi, I am a new user to Proteowizard / msconvert but I've been having issues running Agilent data through msconvert. 
I've tried this using:  msconvertGUI, windows command line, and wine/docker - all three I received the same error (below). 
I've tried this with raw and preprocessed data that I've gotten from a couple different sources. 

Error message:
error: [SAXParser::parse()] Empty tag at offset -1.

The end goal is to dockerize this and include it in a workflow to determine CCS values.

Also I did not see any Agilent raw test data in your repository. This could be helpful for others trying to troubleshoot issues.

nickshulman, 2022-06-03:
Can you send us your raw data? We will figure out what is going wrong.

jjacobson95, 2022-06-03:
Yes, thank you.  This data has been preprocessed using PNNL Preprocessor.

Please let me know if you'd like any more information.

Best,
Jeremy

 Dorrestein_GnPS_P14_G10_7V_Rep01_Neg_29Dec20_Fi...
<https://drive.google.com/file/d/1myyiwulF629Dwed5i6b0GJVj4ZbTbybK/view?usp=drive_web>

 Dorrestein_GnPS_P14_G10_msms_Neg_29Dec20_Fiji_I...
<https://drive.google.com/file/d/1iqW2jGXv7oJtlhKCgpsk_sSMK82tcQEZ/view?usp=drive_web>

 AgTune_01_7V_Neg_02Jan20_Fiji_Infusion_Min50_MA...
<https://drive.google.com/file/d/1-wjRClMRyDLVt33F34BxJ9tEGAApOMZ5/view?usp=drive_web>

 AgTune_01_msms_POS_01Dec20_Fiji_Infusion_Min50_...
<https://drive.google.com/file/d/1jf4enifvLbOXzmI4My5il-68PrXxwJuF/view?usp=drive_web>

nickshulman, 2022-06-03:
msconvert.exe seems to be confused by the line endings in the file "Devices.xml".
That file uses line feeds as line endings ("\n"). If I replace those line endings with CRLF ("\r\n") then it seems to fix things.
I will ask around and see if someone can fix this.
Hope this helps.
-- Nick

PR #2125 (2022-06-16): Fix a few pwiz issues
- added ms levels to processingMethod for SpectrumList_PeakPicker when the levels are not simply "1-"
Useful for understanding how a file was created without analyzing each scan's dataProcessing references

- added scan time and precursor intensity for MS2 "EZ" lines
Requested by @mhoopmann

- fixed error reading MassHunter Devices.xml with LF line endings
Closes #2121
closed 2022-06-16 by chambm