ProteoWizard/pwiz #2151: msconvert.exe raw output mzML differs from proteome-discoverer impacting downstream processing with FLASHDeconv.exe

animesh, 2022-07-07:
I tried the default conversion of a raw file output from Thermo/HF

```
F:\GD\OneDrive\Dokumenter\GitHub\scripts>"F:\OneDrive - NTNU\ProteoWizard 3.0.22155.0ff594f 64-bit\msconvert.exe" "L:\promec\HF\Lars\2022\june\Solveig\220627_26_IgA.raw"
format: mzML
    m/z: Compression-None, 64-bit
    intensity: Compression-None, 32-bit
    rt: Compression-None, 64-bit
ByteOrder_LittleEndian
 indexed="true"
outputPath: .
extension: .mzML
contactFilename:
runIndexSet:

spectrum list filters:

chromatogram list filters:

filenames:
  L:\promec\HF\Lars\2022\june\Solveig\220627_26_IgA.raw

processing file: L:\promec\HF\Lars\2022\june\Solveig\220627_26_IgA.raw
calculating source file checksums

writing output file: .\220627_26_IgA.mzML
```

along with proteome-discoverer 2.5 
[PWF_QE_Basic_mzMLx - Copy.pdProcessingWF.txt](https://github.com/ProteoWizard/pwiz/files/9062636/PWF_QE_Basic_mzMLx.-.Copy.pdProcessingWF.txt)

but the mzML generated by these process seems to have differing results, [msconvert](https://github.com/ProteoWizard/pwiz/files/9062675/220627_26_IgA-MS.txt) and [proteome-discoverer 2.5 ](https://github.com/ProteoWizard/pwiz/files/9062668/220627_26_IgA-PD.txt) respectively  from FLASHDeconv.exe with same configuration, thus wondering if this is a common issue otherwise which one is more or less accurate?

Can share the raw file if the need be?

chambm, 2022-07-07:
Is there a difference in peak picking? The default config will not do any centroiding if the raw data is profile.

animesh, 2022-07-07:
Thanks @chambm , will do so! This data is unpublished but i can share it privately, can you send me an email to animesh.sharma@ntnu.no and i will share the download link 👍

chambm, 2022-07-07:
It's profile data. The 1.8GB vs. 162MB file size difference (both with 2570 scans) should have been a clue! :) Run msconvert with peakPicking and you should get a very similar file.

animesh, 2022-07-08:
Thanks @chambm for looking into this and indeed running with  "peakPicking true 1-"  
```
"F:\OneDrive - NTNU\ProteoWizard 3.0.22155.0ff594f 64-bit\msconvert.exe"  --filter "peakPicking true 1-" "L:\promec\HF\Lars\2022\june\Solveig\220627_26_IgA.raw"
```
did reduce the output to ~160MB and the results are quite similar now
[220627_26_IgA.mzMLfdc - Copy.txt](https://github.com/ProteoWizard/pwiz/files/9069978/220627_26_IgA.mzMLfdc.-.Copy.txt)
 , there is still many difference though, is there way to optimize peak picking further or it is the way it is?

chambm, 2022-07-08:
That's using Thermo's peak picking (which has no parameters), so if the RawFileReader peak picking is different than the PD peak picking, there's not a lot we can do. We do include "exception" peaks now where a couple years ago we didn't. But that should only add peaks, and in some of the differences I see the PD result has more peaks instead of less. If you upload the new mzML I can look at the differences more closely, but the result is probably going to be, as you said, "it is the way it is".

animesh, 2022-07-08:
OK, i've loaded the file and mailed the link. Lets see if something interesting pops up?

chambm, 2022-07-08:
Thanks. I see I had misinterpreted the "Mass count" column in the Flashdeconv output: it doesn't refer to spectrum peak counts. On a spectrum by spectrum basis, the msconvert version always has the same or more peaks. That's almost certainly due to exception peaks, or possibly some kind of extra filtering on the PD mzML. I wish I knew what an "exception peak" really was: assuming that's what they are, your data has an unusually high number of them. Probably need to ask a Thermo guru about it.

closed 2022-07-08 by chambm