ProteoWizard/pwiz #2202: MSConvert outputs erroneous SPS-MS3 precursors

lazear, 2022-08-11:
Hi,

I was testing out some software for analyzing SPS-MS3 data, and I noticed that every single MS3 scan in my mzML file had an extra precursor listed that is completely bogus. For our method, we select 10 ions for SPS. Most MS3 scans in the mzML file list 11 precursors, and the last precursor points to an MS1 scan (unrelated to the MS1-MS2-MS3 chain of the current scan), and lists a selected ion m/z that is not even present in that scan!

Opening the RAW file in Proteome Discoverer shows the 10 correct SPS ions, so I don't believe this is an issue present in the raw data itself. Have I misconfigured MSConvert, or is this a bug?

```
Version: 3.0.22198-0867718 (automated build)

msconvert.exe --32 --zlib --filter "peakPicking vendor msLevel=1-" --filter "scanNumber 16246-16280" --filter "titleMaker <RunId>.<ScanNumber>.<ScanNumber>.<ChargeState> File:"""^<SourcePath^>""", NativeID:"""^<Id^>""""  "D:\Yeast_TMT_TKO_1.raw"
```

For example, the following MS2 scan (Scan 16270) comes from MS1 Scan 16213 (extraneous information removed - happy to share a slice of the mzML file if needed)

```xml
     <spectrum index="16270" id="controllerType=0 controllerNumber=1 scan=16271" defaultArrayLength="707">
        <cvParam cvRef="MS" accession="MS:1000580" name="MSn spectrum" value=""/>
        <cvParam cvRef="MS" accession="MS:1000511" name="ms level" value="2"/>
        <cvParam cvRef="MS" accession="MS:1000130" name="positive scan" value=""/>
        <cvParam cvRef="MS" accession="MS:1000127" name="centroid spectrum" value=""/>
        <cvParam cvRef="MS" accession="MS:1000504" name="base peak m/z" value="497.722045898438" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
        <cvParam cvRef="MS" accession="MS:1000505" name="base peak intensity" value="21016.42578125" unitCvRef="MS" unitAccession="MS:1000131" unitName="number of detector counts"/>
        <cvParam cvRef="MS" accession="MS:1000285" name="total ion current" value="6.998992494049e05"/>
        <cvParam cvRef="MS" accession="MS:1000528" name="lowest observed m/z" value="146.859008789063" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
        <cvParam cvRef="MS" accession="MS:1000527" name="highest observed m/z" value="1005.533996582031" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
        <cvParam cvRef="MS" accession="MS:1000796" name="spectrum title" value="Yeast_TMT_TKO_1.16271.16271.2 File:&quot;Yeast_TMT_TKO_1.raw&quot;, NativeID:&quot;controllerType=0 controllerNumber=1 scan=16271&quot;"/>
        <precursorList count="1">
          <precursor spectrumRef="controllerType=0 controllerNumber=1 scan=16213">
            <isolationWindow>
              <cvParam cvRef="MS" accession="MS:1000827" name="isolation window target m/z" value="506.777069091797" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
              <cvParam cvRef="MS" accession="MS:1000828" name="isolation window lower offset" value="0.25" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
              <cvParam cvRef="MS" accession="MS:1000829" name="isolation window upper offset" value="0.25" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
              <userParam name="ms level" value="1"/>
            </isolationWindow>
            <selectedIonList count="1">
              <selectedIon>
                <cvParam cvRef="MS" accession="MS:1000744" name="selected ion m/z" value="506.777069091797" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                <cvParam cvRef="MS" accession="MS:1000041" name="charge state" value="2"/>
                <cvParam cvRef="MS" accession="MS:1000042" name="peak intensity" value="5.8762082e07" unitCvRef="MS" unitAccession="MS:1000131" unitName="number of detector counts"/>
              </selectedIon>
            </selectedIonList>
            <activation>
              <cvParam cvRef="MS" accession="MS:1000133" name="collision-induced dissociation" value=""/>
              <cvParam cvRef="MS" accession="MS:1000045" name="collision energy" value="30.0" unitCvRef="UO" unitAccession="UO:0000266" unitName="electronvolt"/>
            </activation>
          </precursor>
        </precursorList>
      </spectrum>
```

And here is a daughter MS3 scan, where 10/11 of listed SPS precursors are correct (selected ions from 16270). Note that the 11th SPS precursor points to a completely unrelated MS1 scan... and the selected ion m/z is not present in that scan!

```xml
      <spectrum index="16276" id="controllerType=0 controllerNumber=1 scan=16277" defaultArrayLength="93">
        <cvParam cvRef="MS" accession="MS:1000580" name="MSn spectrum" value=""/>
        <cvParam cvRef="MS" accession="MS:1000511" name="ms level" value="3"/>
        <cvParam cvRef="MS" accession="MS:1000130" name="positive scan" value=""/>
        <cvParam cvRef="MS" accession="MS:1000127" name="centroid spectrum" value=""/>
        <cvParam cvRef="MS" accession="MS:1000504" name="base peak m/z" value="130.134429931641" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
        <cvParam cvRef="MS" accession="MS:1000505" name="base peak intensity" value="1.6702625e05" unitCvRef="MS" unitAccession="MS:1000131" unitName="number of detector counts"/>
        <cvParam cvRef="MS" accession="MS:1000285" name="total ion current" value="2.3480745e06"/>
        <cvParam cvRef="MS" accession="MS:1000528" name="lowest observed m/z" value="112.533386230469" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
        <cvParam cvRef="MS" accession="MS:1000527" name="highest observed m/z" value="489.286407470703" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
        <cvParam cvRef="MS" accession="MS:1000796" name="spectrum title" value="Yeast_TMT_TKO_1.16277.16277. File:&quot;Yeast_TMT_TKO_1.raw&quot;, NativeID:&quot;controllerType=0 controllerNumber=1 scan=16277&quot;"/>
        <userParam name="master scan number" value="16270" type="xsd:positiveInteger"/>
        <scanList count="1">
          <cvParam cvRef="MS" accession="MS:1000795" name="no combination" value=""/>
          <scan>
            <cvParam cvRef="MS" accession="MS:1000016" name="scan start time" value="25.8960031152" unitCvRef="UO" unitAccession="UO:0000031" unitName="minute"/>
            <cvParam cvRef="MS" accession="MS:1000512" name="filter string" value="FTMS + c NSI d Full ms3 606.0185@cid30.00 727.3461@hcd55.00 [110.0000-500.0000]"/>
[... 176 lines cut ...]
            <activation>
              <cvParam cvRef="MS" accession="MS:1000133" name="collision-induced dissociation" value=""/>
              <cvParam cvRef="MS" accession="MS:1000045" name="collision energy" value="55.0" unitCvRef="UO" unitAccession="UO:0000266" unitName="electronvolt"/>
            </activation>
          </precursor>
          <precursor spectrumRef="controllerType=0 controllerNumber=1 scan=16246">
            <isolationWindow>
              <cvParam cvRef="MS" accession="MS:1000827" name="isolation window target m/z" value="606.018493652344" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
              <cvParam cvRef="MS" accession="MS:1000828" name="isolation window lower offset" value="0.25" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
              <cvParam cvRef="MS" accession="MS:1000829" name="isolation window upper offset" value="0.25" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
              <userParam name="ms level" value="1"/>
            </isolationWindow>
            <selectedIonList count="1">
              <selectedIon>
                <cvParam cvRef="MS" accession="MS:1000744" name="selected ion m/z" value="606.018493652344" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
              </selectedIon>
            </selectedIonList>
            <activation>
              <cvParam cvRef="MS" accession="MS:1000133" name="collision-induced dissociation" value=""/>
              <cvParam cvRef="MS" accession="MS:1000045" name="collision energy" value="30.0" unitCvRef="UO" unitAccession="UO:0000266" unitName="electronvolt"/>
            </activation>
          </precursor>
        </precursorList>
      </spectrum>
 ```

chambm, 2022-08-11:
Unfortunately mzML doesn't have an official representation to indicate the full chain of precursors in an MS3+ spectrum. It was designed with the idea that the MS3 spectrum would point at the MS2, which would point at the MS1. So if you had an MS3 and you wanted to know what the MS2 precursor was (pointing at the MS1), you'd go look up the MS2 scan's precursor information. That worked as long as the MS2 was in the file. But mzML didn't anticipate that instruments would be able to drop (or never perform) the MS2 scans from the file at acquisition time. In such files there is no MS2 spectrum to go to to get the MS2 precursor. So in ProteoWizard we add the extra levels of precursor information to the precursorList and specify which `ms level` each precursor refers to with a userParam. We put the highest ms level first so that most reader code that only checks the first precursor gets the most appropriate one. When there are multiple precursors as in SPS, reader code needs to be modified to handle the userParam and drop the ms levels the code isn't interested in (some code might be only interested in the MS1, for example).

lazear, 2022-08-11:
Got it - what I don't understand is why the MS1 scan listed in the MS3 precursor list is different than if I follow the chain from the proceeding 10 precursors up to the MS1 level.

chambm, 2022-08-11:
Can you replace scan 16271 above with 16270 so it corresponds with the MS3?

lazear, 2022-08-11:
Whoops, didn't realize I pasted the wrong one - unfortunately it's not a copy-paste error, I programmatically went through the entire file and checked.

Here's scan 16270 - still has MS1 scan 16213 as the parent, not 16246 as listed in the MS3
```xml
  <spectrum index="16269" id="controllerType=0 controllerNumber=1 scan=16270" defaultArrayLength="672">
        <cvParam cvRef="MS" accession="MS:1000580" name="MSn spectrum" value=""/>
        <cvParam cvRef="MS" accession="MS:1000511" name="ms level" value="2"/>
        <cvParam cvRef="MS" accession="MS:1000130" name="positive scan" value=""/>
        <cvParam cvRef="MS" accession="MS:1000127" name="centroid spectrum" value=""/>
        <cvParam cvRef="MS" accession="MS:1000504" name="base peak m/z" value="886.39501953125" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
        <cvParam cvRef="MS" accession="MS:1000505" name="base peak intensity" value="99222.828125" unitCvRef="MS" unitAccession="MS:1000131" unitName="number of detector counts"/>
        <cvParam cvRef="MS" accession="MS:1000285" name="total ion current" value="2.043994709457e06"/>
        <cvParam cvRef="MS" accession="MS:1000528" name="lowest observed m/z" value="176.120010375977" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
        <cvParam cvRef="MS" accession="MS:1000527" name="highest observed m/z" value="1561.573974609375" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
        <cvParam cvRef="MS" accession="MS:1000796" name="spectrum title" value="Yeast_TMT_TKO_1.16270.16270.3 File:&quot;Yeast_TMT_TKO_1.raw&quot;, NativeID:&quot;controllerType=0 controllerNumber=1 scan=16270&quot;"/>
        <scanList count="1">
          <cvParam cvRef="MS" accession="MS:1000795" name="no combination" value=""/>
          <scan instrumentConfigurationRef="IC2">
            <cvParam cvRef="MS" accession="MS:1000016" name="scan start time" value="25.883510178667" unitCvRef="UO" unitAccession="UO:0000031" unitName="minute"/>
            <cvParam cvRef="MS" accession="MS:1000512" name="filter string" value="ITMS + c NSI r d Full ms2 606.0185@cid30.00 [162.0000-1829.0000]"/>
            <cvParam cvRef="MS" accession="MS:1000927" name="ion injection time" value="2.605605840683" unitCvRef="UO" unitAccession="UO:0000028" unitName="millisecond"/>
            <userParam name="[Thermo Trailer Extra]Monoisotopic M/Z:" value="606.01849365234375" type="xsd:float"/>
            <scanWindowList count="1">
              <scanWindow>
                <cvParam cvRef="MS" accession="MS:1000501" name="scan window lower limit" value="162.0" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                <cvParam cvRef="MS" accession="MS:1000500" name="scan window upper limit" value="1829.0" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
              </scanWindow>
            </scanWindowList>
          </scan>
        </scanList>
        <precursorList count="1">
          <precursor spectrumRef="controllerType=0 controllerNumber=1 scan=16213">
            <isolationWindow>
              <cvParam cvRef="MS" accession="MS:1000827" name="isolation window target m/z" value="606.018493652344" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
              <cvParam cvRef="MS" accession="MS:1000828" name="isolation window lower offset" value="0.25" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
              <cvParam cvRef="MS" accession="MS:1000829" name="isolation window upper offset" value="0.25" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
              <userParam name="ms level" value="1"/>
            </isolationWindow>
            <selectedIonList count="1">
              <selectedIon>
                <cvParam cvRef="MS" accession="MS:1000744" name="selected ion m/z" value="606.018493652344" unitCvRef="MS" unitAccession="MS:1000040" unitName="m/z"/>
                <cvParam cvRef="MS" accession="MS:1000041" name="charge state" value="3"/>
                <cvParam cvRef="MS" accession="MS:1000042" name="peak intensity" value="2.3733592e07" unitCvRef="MS" unitAccession="MS:1000131" unitName="number of detector counts"/>
              </selectedIon>
            </selectedIonList>
            <activation>
              <cvParam cvRef="MS" accession="MS:1000133" name="collision-induced dissociation" value=""/>
              <cvParam cvRef="MS" accession="MS:1000045" name="collision energy" value="30.0" unitCvRef="UO" unitAccession="UO:0000266" unitName="electronvolt"/>
            </activation>
          </precursor>
        </precursorList>
      </spectrum>
```

lazear, 2022-08-11:
Interesting that the MS3 precursor scan listed (16243) and the MS2 precursor scan listed (16213) have the same selected ion m/z - so is this just a mislabeling somewhere?

chambm, 2022-08-11:
Hmm, that's probably because in the MS3 it only knows the MS2 precursor from the filter line. So to get a precursor scan number, it looks for the first preceding scan with ms level - 2 (for the MS3 -> MS1 case) where the isolation m/z is within the scan range. That can definitely be different from the Master Scan which is the actual scan that triggered the dependent scan. But they will have the same isolation m/z, as you noticed. So if the MS2s are really in the file, you can prefer to look up the MS2 scan and get the precursor info from there. But the MS2s might not be there at all, in which case the ms level - 1 precursors won't even have precursor scan numbers (as long as there are truly no MS2s in the file), and the Master Scan parameter for the MS3 should actually point at the MS1 I think (although I'm not sure we're actually using the Master Scan parameter in that case).

lazear, 2022-08-11:
I'm still somewhat confused I think - when I discovered the issue, all MS level scans are present in the mzML file.

The MS1 scan listed as the precursor (16246) doesn't contain selected ion m/z (606.0185). The two closest m/z peaks are [606.0171, 606.35095].

chambm, 2022-08-12:
606.0185 is pretty close to 606.0171 and could easily be the difference between the isolation (filter line) m/z and the monoisotopic m/z. (Does that exact m/z occur in scan 16213?) Also, do Thermo instruments still do that thing where they take a lower resolution preliminary scan to find dependent targets to trigger (which gets used as the filter line m/z), and then when the high resolution MS1 scan finishes the data is replaced?

Note that the pwiz precursor scan number heuristic for MS3->MS1 doesn't check that the isolation m/z is present in the peak list (that would be too slow), it only checks that it is in the scan range (i.e. between the min and max observed m/z). Also the "selection ion m/z" for MS3->MS1 is just going to be a copy of the isolation m/z; it can't get the monoisotopic m/z for a scan 2 MS levels away directly from the Thermo API.

Basically the MS1 precursor is only there in the MS3 in case the MS2s are not even in the file (or in case software wants a faster way to get the MS1 isolation m/z without having to look up another scan).

lazear, 2022-08-13:
> Note that the pwiz precursor scan number heuristic for MS3->MS1 doesn't check that the isolation m/z is present in the peak list (that would be too slow), it only checks that it is in the scan range [...]

Got it!

Well, I will just close the issue, and ignore the non-MS2 precursor scans on my side. Thanks for answering my questions - and for maintaining MSConvert!

closed 2022-08-13 by lazear