ProteoWizard/pwiz #2253: openshift (no sudo/root access) ready image?

kellenrl, 2022-08-30:
Hi, I am an HPC Engineer at the Oak Ridge Leadership Computing Facility and we are attempting to run ProteoWizard on our OpenShift kubernetes clusters for a project. OpenShift does not allow containers to have root access, and we are hitting the following errors when attempting to launch the image (https://hub.docker.com/r/chambm/pwiz-skyline-i-agree-to-the-vendor-licenses) into a pod:

```
sudo: PERM_SUDOERS: setresuid(-1, 1, -1): Operation not permitted
sudo: no valid sudoers sources found, quitting
sudo: setresuid() [0, 0, 0] -> [2002244, -1, -1]: Operation not permitted
sudo: unable to initialize policy plugin
```

We were wondering if you had - or would be willing to create - an image that can be used without root access. Any help or insight is greatly appreciated.

chambm, 2022-08-30:
There is a mywine wrapper in there that will copy the WINEPREFIX to a new directory owned by whatever user you're running the docker command as. Then it will use that WINEPREFIX to run the wine command.

jspaezp, 2022-09-22:
@chambm would that entail calling it as "docker run -it --rm chambm/pwiz-skyline-i-agree-to-the-vendor-licenses mywine msconvert --help" instead of "docker run -it --rm chambm/pwiz-skyline-i-agree-to-the-vendor-licenses wine msconvert --help" ??

chambm, 2022-10-18:
Yes.