ProteoWizard/pwiz #2270: Sciex 7500 .wiff to mzML losing transition information

PMSeitzer, 2022-09-08:
Hello,

I am using Proteowizard msconvert from version `3.0.22244-e838899 (automated build)` on a windows 10 VM, trying to convert sciex .wiff files to mzML.  The instrument is a Sciex 7500 QQQ.

Here is the command I used:
```
msconvert.exe <input_file> --64 --filter "peakPicking 1-" --mzML -o <output_file> --ignoreUnknownInstrumentError
```

It's worth mentioning that this `--ignoreUnknownInstrumentError` flag is necessary, otherwise the conversion fails.

The conversion is successful, but the resulting data in the mzML has some errors in it. For one thing, the transition information (the compound Id) is missing. It is replaced by a serial number.  Looking over a series of runs,  I can see that this serial number is random, and cannot be used to map back to transitions reliably - e.g., `transition=17` might be `transition=23` in a different file.
The `precursorMz` and `productMz` (e.g., `Q1` and `Q3`) info does make it in to the `srmId` string, however, this isn't enough information for me to identify my transitions, b/c I have identical `(precursorMz, productMz)` values with different collision energies and RT ranges.  The collision energy information is also not carried along through the conversion.

Here is a sample of some of the converted file information:

```
       <chromatogram index="2" id="SRM SIC Q1=170.12 Q3=109.1 sample=2 period=1 experiment=1 transition=0" defaultArrayLength="2019">
          <cvParam cvRef="MS" accession="MS:1001473" name="selected reaction monitoring chromatogram" value=""/>
          <cvParam cvRef="MS" accession="MS:1000130" name="positive scan" value=""/>
          <userParam name="MS_dwell_time" value="0.0030000000000000001" type="xs:float"/>
          <precursor>
            <isolationWindow>

 <cvParam cvRef="MS" accession="MS:1000045" name="collision energy" value="0.0" unitCvRef="UO" unitAccession="UO:0000266" unitName="electronvolt"/>
        
```
Note the `transition=0` serial number, and that `MS:1000045` shows a collision energy value of `0.0`, which is not correct.

Happy to provide raw files if that would be helpful in debugging this issue.

PMSeitzer, 2022-09-08:
For reference, here is what I hoped to see:
```
       <chromatogram index="2" id="SRM SIC Q1=170.12 Q3=109.1 sample=2 period=1 experiment=1 transition=My Favorite Compound Transition 2" defaultArrayLength="2019">
          <cvParam cvRef="MS" accession="MS:1001473" name="selected reaction monitoring chromatogram" value=""/>
          <cvParam cvRef="MS" accession="MS:1000130" name="positive scan" value=""/>
          <userParam name="MS_dwell_time" value="0.0030000000000000001" type="xs:float"/>
          <precursor>
            <isolationWindow>

 <cvParam cvRef="MS" accession="MS:1000045" name="collision energy" value="20.0" unitCvRef="UO" unitAccession="UO:0000266" unitName="electronvolt"/>
```
Note the collision energy value of `20 EV` and the transition Id string is now `my favorite compound transition 2` instead of a random serial number.

chambm, 2022-09-08:
Hi, please do provide a wiff and wiff.scan file so I can check it out. It looks like we used to record CE and DP info but I forgot to re-implement it after we changed to a newer Sciex data access API. The transition field is a direct index into the API's MassRangeInfo for SIM/SRM transitions, so it's not random. I'm not sure your `my favorite compound transition string 2` is available in the API, but I can check.

PMSeitzer, 2022-09-08:
Hey @chambm , thanks for responding. I've invited you to share a google drive folder using the gmail address listed on your github profile.  Let me know if you have any problem accessing the data.

chambm, 2022-09-08:
You mentioned your RT ranges are different for the same Q1/Q3 but the file you shared appears to be all unscheduled MRM samples. Was that a different file?

PMSeitzer, 2022-09-08:
Hey @chambm , sorry , that data is unscheduled. What differs are the collision energies.  I think if you were able to pass the collision energy information through, that would be enough information to distinguish transitions from each other.  Essentially, I'd be able to define unique `(precursorMz, productMz, CE)` keys for each compound ahead of time, then read them out of the mzML files. 

However, if you were able to pass along the transition name instead of `transition=X` into the `id` field, that would be hugely helpful.

Glad to know the transition id number isn't random.  I did a quick test and found that in different samples, the transition id for the same unique `(precursorMz, productMz)` pair was assigned a different value - so the randomness may be in how the instrument decides to collect data for transitions, and generate the index.  If the mapping was consistent across samples and runs, the transition id number could be used to map transitions to the compound information, even though that information isn't currently passed along to the mzMLs.

closed 2022-09-09 by chambm