ProteoWizard/pwiz #2277: Sciex 7500 SRM Ids missing during file conversion

PMSeitzer, 2022-09-09:
This issue is a follow-up to #2272 , which was graciously fixed by @chambm 

In the file conversion process from .wiff files generated by Sciex 7500 for MRM mode, it appears that many SRM Ids that had previously been retained are no longer available.

Here is a list of the SRM IDs collected from recent Proteowizard build `3.0.22252 f9ce728`:

```
- SRM SIC Q1=279.2 Q3=163.1 sample=1 period=1 experiment=1 transition=4 ce=22 name=12-HHT (279/163)
- SRM SIC Q1=295.3 Q3=195.2 sample=3 period=1 experiment=1 transition=0 ce=25 name=12(13)-EpOME (295/195)
- SRM SIC Q1=315.2 Q3=271.1 sample=19 period=1 experiment=1 transition=10 ce=20 name=15-deoxy-PGJ2 (315/271)
- SRM SIC Q1=319.2 Q3=115.1 sample=1 period=1 experiment=1 transition=9 ce=21 name=5-HETE (319/115)
- SRM SIC Q1=319.2 Q3=179.1 sample=1 period=1 experiment=1 transition=8 ce=21 name=12-HETE (319/179)
- SRM SIC Q1=319.2 Q3=219.1 sample=1 period=1 experiment=1 transition=6 ce=19 name=15-HETE (319/219)
- SRM SIC Q1=327.2 Q3=283.1 sample=17 period=1 experiment=1 transition=6 ce=18 name=DHA (327/283)
- SRM SIC Q1=333.2 Q3=271.1 sample=19 period=1 experiment=1 transition=8 ce=22 name=PGJ2 (333/271)
- SRM SIC Q1=343.2 Q3=101.1 sample=16 period=1 experiment=1 transition=0 ce=17 name=(±)4-HDHA (343/101)
- SRM SIC Q1=343.2 Q3=141.1 sample=16 period=1 experiment=1 transition=1 ce=18 name=(±)7-HDHA (343/141)
- SRM SIC Q1=343.2 Q3=193.1 sample=16 period=1 experiment=1 transition=6 ce=17 name=(±)13-HDHA (343/193)
- SRM SIC Q1=343.2 Q3=233 sample=16 period=1 experiment=1 transition=8 ce=20 name=(±)16-HDHA (343/233)
- SRM SIC Q1=349.2 Q3=113.1 sample=19 period=1 experiment=1 transition=3 ce=28 name=15-keto-PGE2 (349/113)
- SRM SIC Q1=351.2 Q3=113.1 sample=19 period=1 experiment=1 transition=1 ce=35 name=15-keto PGF2a (351/113)
- SRM SIC Q1=351.2 Q3=189.1 sample=1 period=1 experiment=1 transition=3 ce=25 name=PGE2 and PGD2 (351/189)
- SRM SIC Q1=353.2 Q3=193.1 sample=19 period=1 experiment=1 transition=0 ce=30 name=11-beta-PGF2a (353/193)
- SRM SIC Q1=353.2 Q3=193.1 sample=1 period=1 experiment=1 transition=2 ce=34 name=PGF2a (353/193)
- SRM SIC Q1=359.2 Q3=199 sample=17 period=1 experiment=1 transition=3 ce=21 name=RvD5 (359/199)
- SRM SIC Q1=369.2 Q3=163.1 sample=1 period=1 experiment=1 transition=0 ce=35 name=6k-PGF1a (369/163)
- SRM SIC Q1=369.2 Q3=169.1 sample=1 period=1 experiment=1 transition=1 ce=22 name=TXB2 (369/169)
- SRM SIC Q1=375.2 Q3=141.1 sample=17 period=1 experiment=1 transition=5 ce=21 name=17(R)-RvD1 (375/141)
- SRM SIC Q1=375.2 Q3=141.1 sample=17 period=1 experiment=1 transition=0 ce=21 name=RvD1 (375/141)
- SRM SIC Q1=375.2 Q3=147 sample=17 period=1 experiment=1 transition=2 ce=25 name=RvD3 (375/147)
- SRM SIC Q1=375.2 Q3=175.1 sample=17 period=1 experiment=1 transition=1 ce=30 name=RvD2 (375/175)
```

Here is a list of SRM IDs collected from an older build:
```
- SRM SIC Q1=279.2 Q3=163.1 sample=1 period=1 experiment=1 transition=27
SRM SIC Q1=282.4 Q3=62.1 sample=1 period=1 experiment=2 transition=29
- SRM SIC Q1=293.2 Q3=113.2 sample=1 period=1 experiment=1 transition=118
- SRM SIC Q1=293.2 Q3=195 sample=1 period=1 experiment=1 transition=121
- SRM SIC Q1=293.3 Q3=171.1 sample=1 period=1 experiment=1 transition=120
- SRM SIC Q1=295.2 Q3=171.1 sample=1 period=1 experiment=1 transition=111
- SRM SIC Q1=295.2 Q3=195.1 sample=1 period=1 experiment=1 transition=112
- SRM SIC Q1=295.3 Q3=195.2 sample=1 period=1 experiment=1 transition=117
- SRM SIC Q1=295.5 Q3=171.1 sample=3 period=1 experiment=1 transition=57
SRM SIC Q1=300.4 Q3=62.1 sample=1 period=1 experiment=2 transition=20
- SRM SIC Q1=301.2 Q3=257.1 sample=1 period=1 experiment=1 transition=79
- SRM SIC Q1=303.2 Q3=259.1 sample=1 period=1 experiment=1 transition=59
- SRM SIC Q1=315.2 Q3=271.1 sample=1 period=1 experiment=1 transition=21
- SRM SIC Q1=317.2 Q3=115.1 sample=1 period=1 experiment=1 transition=66
- SRM SIC Q1=317.2 Q3=167.1 sample=1 period=1 experiment=1 transition=67
- SRM SIC Q1=317.2 Q3=179.1 sample=1 period=1 experiment=1 transition=68
- SRM SIC Q1=317.2 Q3=219.1 sample=1 period=1 experiment=1 transition=69
- SRM SIC Q1=317.2 Q3=259.1 sample=1 period=1 experiment=1 transition=70
- SRM SIC Q1=319.2 Q3=115.1 sample=1 period=1 experiment=1 transition=50
- SRM SIC Q1=319.2 Q3=167.1 sample=1 period=1 experiment=1 transition=44
- SRM SIC Q1=319.2 Q3=179.1 sample=1 period=1 experiment=1 transition=53
- SRM SIC Q1=319.2 Q3=219.1 sample=1 period=1 experiment=1 transition=45
- SRM SIC Q1=319.2 Q3=231.1 sample=1 period=1 experiment=1 transition=56
- SRM SIC Q1=319.2 Q3=261.2 sample=1 period=1 experiment=1 transition=55
- SRM SIC Q1=319.2 Q3=275.1 sample=10 period=1 experiment=1 transition=1
- SRM SIC Q1=319.2 Q3=289.1 sample=1 period=1 experiment=1 transition=57
- SRM SIC Q1=319.3 Q3=167.2 sample=1 period=1 experiment=1 transition=51
- SRM SIC Q1=321.4 Q3=195.3 sample=1 period=1 experiment=1 transition=119
SRM SIC Q1=322.4 Q3=62.1 sample=1 period=1 experiment=2 transition=28
- SRM SIC Q1=324.2 Q3=277.1 sample=1 period=1 experiment=1 transition=113
SRM SIC Q1=324.4 Q3=62.1 sample=1 period=1 experiment=2 transition=23
- SRM SIC Q1=326.2 Q3=168.1 sample=1 period=1 experiment=1 transition=114
- SRM SIC Q1=326.2 Q3=169.1 sample=1 period=1 experiment=1 transition=115
SRM SIC Q1=326.4 Q3=62.1 sample=1 period=1 experiment=2 transition=19
- SRM SIC Q1=327.2 Q3=116.1 sample=1 period=1 experiment=1 transition=3
[... 62 lines cut ...]
- SRM SIC Q1=380.2 Q3=141.1 sample=1 period=1 experiment=1 transition=7
- SRM SIC Q1=380.2 Q3=152.1 sample=1 period=1 experiment=1 transition=8
SRM SIC Q1=396.3 Q3=62.1 sample=1 period=1 experiment=2 transition=17
SRM SIC Q1=398.3 Q3=62.1 sample=1 period=1 experiment=2 transition=18
SRM SIC Q1=440.3 Q3=189.1 sample=1 period=1 experiment=2 transition=6
SRM SIC Q1=445.2 Q3=194.1 sample=1 period=1 experiment=2 transition=2
SRM SIC Q1=464.3 Q3=191.1 sample=1 period=1 experiment=2 transition=12
SRM SIC Q1=464.3 Q3=231.1 sample=1 period=1 experiment=2 transition=15
SRM SIC Q1=497.3 Q3=189.1 sample=1 period=1 experiment=2 transition=5
SRM SIC Q1=502.3 Q3=194.1 sample=1 period=1 experiment=2 transition=1
- SRM SIC Q1=508.3 Q3=59.1 sample=1 period=1 experiment=1 transition=122
SRM SIC Q1=521.3 Q3=191.1 sample=1 period=1 experiment=2 transition=11
SRM SIC Q1=521.3 Q3=231.1 sample=1 period=1 experiment=2 transition=14
SRM SIC Q1=524.3 Q3=183.9 sample=1 period=1 experiment=2 transition=16
SRM SIC Q1=626.3 Q3=189.1 sample=1 period=1 experiment=2 transition=4
SRM SIC Q1=631.3 Q3=194.1 sample=1 period=1 experiment=2 transition=0
SRM SIC Q1=650.3 Q3=191.1 sample=1 period=1 experiment=2 transition=10
SRM SIC Q1=650.3 Q3=231.1 sample=1 period=1 experiment=2 transition=13
```

It looks like there are many fewer files from the conversion also:

old Proteowizard:
```
```

PMSeitzer, 2022-09-09:
I wonder if the file contains a mixture of scheduled and unscheduled MRMs, and the new converter is only retaining unscheduled MRMs?

chambm, 2022-09-12:
Are you sure you're comparing the same WIFF files? The first part of the mzML output is the WIFF filename which should definitely be the same between the new and old versions.

PMSeitzer, 2022-09-14:
Hey @chambm thanks for responding. The analyst that generated those files is currently out of office, but I'll follow up with them and report back.  At first glance, it does appear that the WIFF files may have somehow gotten mixed up.

PMSeitzer, 2022-10-06:
@chambm I looked into it, and I think this was just a case of user error. Feel free to close this issue whenever you'd like.

closed 2022-10-06 by PMSeitzer