ProteoWizard/pwiz #2287: MSconvert fails to get the calibrated mass

MuyaoXi9271, 2022-09-16:
Hi expert,

The DDA raw data is acquired from Bruker instrument. Then I calibrate the data in order to get the accurate masses by software "dataanalysis" developed by Bruker. Then I export the raw data (.d) and transfer .d into .mzML by using MSconvert. 

When I check the converted file, the precursor before doing calibration is not replaced by the correct precursor after doing calibration.

I am very pleased to send the file and masses for checking to you in order to figure the problem out. 

I am looking forward to receiving any feedback and help from you.

Bests,
Muyao

chambm, 2022-09-16:
Is that BAF data or TIMS?

MuyaoXi9271, 2022-09-16:
I find a file named analysis.baf. So I assume it is the BAF data.

Bests,
Muyao

chambm, 2022-09-16:
OK, I'll ask the Bruker devs about calibration of metadata. I'm not sure it's supposed to be included in the calibration (unfortunately).

chambm, 2022-09-16:
I expect it's too big to send over email, but I'll ask for you to upload it somewhere if the Bruker devs say that the metadata SHOULD be calibrated.

MuyaoXi9271, 2022-09-16:
Yes, it is very big. I am very looking forward to receiving your reply. Have a lovely weekend.

Bests,
Muyao

chambm, 2022-09-19:
The reply I got back from Bruker: 

> Yes, this is a known issue. Just as in the tdf format precursors are stored as m/z values with the “raw” instrument calibration. (In .tdf it is the precursors table)
> 
> The calibration function is just applied to the raw scans – which are stored as time-of-flight values in the .bin files. If we would apply it to the precursor m/z values, we would change the raw data (or the metadata of the raw data).

So what we need is a standalone function that can calibrate individual m/z values on the fly. We don't have that right now.

MuyaoXi9271, 2022-09-19:
Thanks for letting me know. Will MSconvert add this function in the later version?

Bests,
Muyao

chambm, 2022-09-19:
That will depend on whether Bruker adds the function to their API. I'll let you know if they do.

MuyaoXi9271, 2024-08-01:
I noticed the raw files (.d) generated by Bruker were converted into .mzML by msconvert, then the calibrated precursors could be obtained. Please check whether Bruker has added that function to their API. I appreciate any help you can provide.

Best,
Muyao