ProteoWizard/pwiz #2306: [Request] Improved documentation to build standalone BiblioSpec

jspaezp, 2022-09-27:
Hello!

First off, amazing work!

I was following the instructions in here to build a standalone version of BiblioSpec for linux:
https://skyline.ms/wiki/home/software/BiblioSpec/page.view?name=Download%20and%20build

And noticed that the documentation does not represent the current version of the project.

I am attaching here what I got and how it differs from the documented website

```bash
git clone --depth 1 https://github.com/ProteoWizard/pwiz.git
# Docs say trunk/pwiz ... not accurate
cd pwiz

bash ./quickbuild.sh -j2 --abbreviate-paths --hash optimization=space address-model=64 pwiz pwiz_tools/BiblioSpec executables gcc
# Docs say `quickbuild.sh -j4 --hash optimization=space address-model=64 pwiz_tools/BiblioSpec`, I am not sure what the implications of the 'executables gcc' at the end are, but they made it work in my system (copied them from the CICD pipeline in gh actions) 

# and finally it says "The resulting build will be located in the build-<os>-<architecture>/pwiz_tools/BiblioSpec directory."
# Which is not accurate, since that directory contains only the src and tests directories.

# On the other hand these are the locations of the binaries
pwiz/build-linux-x86_64/gcc-release-x86_64/BlibBuild
pwiz/build-linux-x86_64/gcc-release-x86_64/BlibFilter
pwiz/build-linux-x86_64/gcc-release-x86_64/BlibSearch
pwiz/build-linux-x86_64/gcc-release-x86_64/BlibToMs2

# As well as:
./build-linux-x86_64/BiblioSpec/bibliospec-bin-linux-gcc6-release-3_0.tar.bz2

# which when uncompressed ...
./build-linux-x86_64/BiblioSpec/BlibBuild
./build-linux-x86_64/BiblioSpec/BlibFilter
./build-linux-x86_64/BiblioSpec/BlibSearch
./build-linux-x86_64/BiblioSpec/BlibToMs2
```

I hope it helps to fix that!

in a related issue, would it be of interest to you to release compiled versions of bilbiospec for linux/mac?
I understand the overhead of supporting them but I was wondering if there is any other reason.

Kindest wishes,
Sebastian

chambm, 2022-09-27:
Thanks for the review of the documentation. I'll try to get those updates integrated.

All compiled version downloads go through our website, but I just realized we don't have a section for BiblioSpec. Just pwiz and Bumbershoot. I may fix that.

jspaezp, 2022-09-29:
It would be absolutely amazing to have a linux release of bibliospec :)

chambm, 2022-09-29:
In the meantime you can use this:
https://teamcity.labkey.org/viewLog.html?buildId=lastSuccessful&buildTypeId=ProteoWizard_BiblioSpecLinuxX8664subset&tab=artifacts&guest=1

jspaezp, 2022-09-30:
In a related matter:

` --best-scoring ` is not documented in the command line tool (more accurately is documented as a placeholder description).

https://github.com/ProteoWizard/pwiz/blob/5b2ee774d52dc52325e0f14c156bcf2aa4f25f4e/pwiz_tools/BiblioSpec/src/BlibFilter.cpp#L171-L173

Thanks a lot for the link to the build!