ProteoWizard/pwiz #2335: Unable to locate package pwiz-setup-3.0.22298.76c82e9-x86_64.msi

annapurnap2007, 2022-10-25:
I am very new to Ubuntu system and trying to install pwiz. It's giving the error attached. 
[image]
Appreciate any help with this issue.

closed 2022-10-25 by chambm

annapurnap2007, 2022-10-25:
Thanks Chambm! I downloaded Linux  non-vendor native builds and when tried to install, It's giving the error attached.
[image]

Please advice me on how to proceed further.

chambm, 2022-10-25:
That's a tarball (*nix-style zip file). You don't install it with apt.
https://linuxize.com/post/how-to-extract-unzip-tar-gz-file/

annapurnap2007, 2022-10-25:
Thank very much! File extracted, but I don't see any configure file in the list.
[image]

chambm, 2022-10-25:
What are you trying to do with ProteoWizard? All the binaries are there. msconvert, idconvert, msaccess, etc.

annapurnap2007, 2022-10-25:
I have no idea. I downloaded pwiz-bin-linux-x86_64-gcc7-release-3_0_22298_120f8aa.tar.bz2 using option "Linux non-vendor native builds". After extracting, all these components listed.

chambm, 2022-10-25:
Why are you installing/downloading ProteoWizard?

annapurnap2007, 2022-10-25:
I would like to use MSconvert to convert LC-MS RAW files to mzML files. As I mentioned I am new to Linux system. Please advice me on how to proceed. Please share step by step protocol, if you have any.

chambm, 2022-10-25:
You can't use the native non-vendor builds for **vendor** proprietary RAW files. You have to use the Docker image, or switch to Windows and do the conversions there. I highly recommend the Windows route since you seem more comfortable with GUI programs.

annapurnap2007, 2022-10-26:
Thanks Chamber! It's working fine on Windows system.