ProteoWizard/pwiz #2349: Skyline on mono

jspaezp, 2022-10-31:
Hello!

I was wondering whether we could use this thread to start discussing the intentions/requirements to make skyline run on mono (even if it was a restricted non-vendor format version of it).

I recall that the idea was "mentioned" (not really considered) back in the day during ASMS san diego by @brendanx67 and was wondering whether the idea had been floating around.

I am aware that the skyline cli can be used from the docker image (which in my experience is still ... 'temperamental' but extremely useful). But many workflows require having a template file that can only be generated (as far as I know) from the GUI.

Once again thank you very much for working in this amazing piece/series of software!
Sebastian

brendanx67, 2022-10-31:
So, you are hoping for a version of Skyline where the UI could be run on
Linux? If so, we are still not devoting any effort to that. Our answer for
Skyline GUI on Mac and Linux is a Windows VM (or dual boot on a Mac).

We see the Docker container as our answer for the Skyline CLI on Linux
because we believe a lot of computational power for running command-line
processing pipelines is Linux-only. The Docker container makes it possible
for Skyline and MSConvert to be included in these processing pipelines
without needing to transition to Windows OS servers.

chambm, 2022-10-31:
If somebody wanted to contribute this capability, they'd need to provide an alternate version of MsDataFileImpl to be able to read open formats like mzML. The current MsDataFileImpl uses ProteoWizard for file reading and uses C++/CLI to access native code from managed code and managed code from native code. Since C++/CLI is not supported/compiled by gcc/mono, there's no chance of getting that working there. But the rest of Skyline is pretty much pure managed C#.

The Docker/Wine combo gets a bit temperamental around Halloween because Wine works by magic and the border between the real and magical dimensions gets thinner. ;)

jspaezp, 2022-10-31:
Hello @brendanx67 and @chambm,

Thank you very much for the fast responses! That does make very clear the technical implications of the "request".
I guess I will keep using windows as a +25gb bootloader for skyline :P

Kindest wishes,
Sebastian

closed 2023-02-09 by chambm