ProteoWizard/pwiz #2375: [BUG] Misplaced modification mass with bibliospec

jspaezp, 2022-11-08:
Hello there!

I wanted to report some unexpected behavior ...
When using bibliospec, if the peptide follows this patten "n[MODMASS]X[MODMASS]" (note the low caps n),
the mass of both modifications gets added and placed in the first aminoacid (which is now high caps).

I think the expected behaviors would be either of:
1. Error due to the pressence of a low caps AA
2. Keep both separate modifications and AAs
3. Combine both modifications, but discard the first "AA" (since low caps n is the output of percolator denoting terminal mods).

```shell
$ cat bugreport.ssl 
file	scan	charge	sequence	score-type	score
03922_GB8-TMT_TUM_lysn_20_01_01-TMT1-1h-R1.mzML	34293	2	n[229.1629]K[229.1629]AAAAAAALQA	PERCOLATOR QVALUE	1.4581382856704295e-06
03922_GB8-TMT_TUM_lysn_20_01_01-TMT1-1h-R1.mzML	34297	2	n[229.1629]K[229.1629]AAAAAAALQA	PERCOLATOR QVALUE	1.4581382856704295e-06
$ sqlite3 bugreport.blib "SELECT peptideModSeq from RefSpectra" 
N[+458.3]KAAAAAAALQA
N[+458.3]KAAAAAAALQA

# Oddly enough, when using the high caps version, the modification sites are maintained

$ cat bugreport2.ssl 
file	scan	charge	sequence	score-type	score
03922_GB8-TMT_TUM_lysn_20_01_01-TMT1-1h-R1.mzML	34293	2	N[229.1629]K[229.1629]AAAAAAALQA	PERCOLATOR QVALUE	1.4581382856704295e-06
03922_GB8-TMT_TUM_lysn_20_01_01-TMT1-1h-R1.mzML	34297	2	N[229.1629]K[229.1629]AAAAAAALQA	PERCOLATOR QVALUE	1.4581382856704295e-06
$ sqlite3 bugreport2.blib "SELECT peptideModSeq from RefSpectra"
N[+229.2]K[+229.2]AAAAAAALQA
N[+229.2]K[+229.2]AAAAAAALQA
```

LMK what you think.
-Sebastian

chambm, 2022-11-08:
BiblioSpec doesn't support terminal mods, so they are moved to the first or last amino acid. When you make it upper case, it's then treated as an amino acid instead of a "n-terminus marker".

chambm, 2022-11-08:
Oh, but I see it looks like it's including N in the peptide for the first example. Let me check that.

jspaezp, 2022-11-08:
Yeah, I noticed not supporting the terminal mods (more accurately, I was aware of it but my script did not modify the sequences the way I intended it to).

I think the correct handling here would be to have an error and let the user know that the 'n' aminoacid does not exist. let me know what you think!

Thanks a lot for the fast reply!

PR #2378 (2022-11-21): BiblioSpec SSL peptide validation
- added error when SSL peptide sequence has non-uppercase letters or unsupported modification formats (reported by Sebastian)

Closes #2375
closed 2022-11-21 by brendanx67