ProteoWizard/pwiz #2381: automated generation of label-free quantification report of MS/MS

martinPasen, 2022-11-11:
Hello!

I am interested in automating the process of generating the quantification report. I have to process multiple measurements and I want to avoid the GUI. The resulting script would perform:

.raw + .ssl + (.fasta) + config -->  report

where report is for example .csv (but on the format does not matter) and it would contain information such as peak heights and peak retention time/scan number belonging to every identified peptide.

In the optimal case i would want to run it on Linux.

I know that it can be done in Skyline using GUI. Can it be done through terminal/API ?

Thank you very much for all the work on the project!
Martin

brendanx67, 2022-11-11:
You can learn more about the Skyline command-line interface by running
SkylineRunner.exe or SkylineCmd.exe without any command-line arguments.
The same information can be found in the Skyline user interface at Help >
Documentation > Command-Line.

It can be run on Linux from the ProteoWizard docker container distribution.

I don't know of any other option for what you are suggesting, but this
should be a pretty good solution for you.

martinPasen, 2022-11-14:
That is exactly what i was looking for. Thanks for the fast response! 

For people that might want to do the same thing:
When i googled "proteoWizard docker container distribution linux" the first results was "biocontainers/proteowizard". Do not use that (it seems to be outdated), use the website designated to the project:
https://proteowizard.sourceforge.io/download.html

Once again, thank you for the good work :)
Martin

closed 2022-11-14 by martinPasen