ProteoWizard/pwiz #2387: Docker to Singularity for HPC and permissions

tracychew, 2022-11-17:
Hi, I am using a Linux HPC system to use msconvert to convert .WIFF files to .mzML. We can only use Singularity on HPC, and do not have root access. I am using the docker container provided by pwiz and converted it to singularity, however I get the issue described here: https://stackoverflow.com/questions/73328706/running-singularity-container-without-root-as-different-user-inside 

Is there an alternate workaround that does not involve rebuilding the container for every user? We would like to include this as part of a nextflow workflow, and this issue really limits its ease of portability. Thanks!

jspaezp, 2022-11-19:
Ive struggled with this in the past... what I found works for me is ...
documented here:
https://github.com/jspaezp/elfragmentador-data#setting-up-msconvert-on-singularity-

sneumann, 2022-11-22:
Hi, I am using it in a script through
```
for FILE in "${FILES[@]}" ; do
  TMPMYWINEPREFIX=`mktemp -d /dev/shm/wineXXX`
  singularity exec \
    --cleanenv \
    -B "$OUTDIR":/data \
    -B $TMPMYWINEPREFIX:/mywineprefix \
    --writable-tmpfs \
    "$IMAGE" \
    mywine msconvert \
    --32 --zlib --filter "peakPicking true 1-" --filter "zeroSamples removeExtra" \
    --outdir  z:/data/mzML \
    "z:/data/$FILE"
   sleep 2 ; rmdir $TMPMYWINEPREFIX
done
```
So the temporary directory for the wine stuff is created as a user, and then mounting that directory into the container. I haven't tested recently, but this could get you on track. 
Yours, 
Steffen

jonasscheid, 2022-12-15:
Hi @tracychew! I have the exact same issue here. We also want to integrate this docker container in our nextflow pipeline and run it on an HPC. Any progress on this?

I would be super grateful for a smooth solution