ProteoWizard/pwiz #2407: Difference in Results between MSConvert GUI & MSConvert CMD

imeMFK01, 2022-12-05:
Hi, 

I am using MSConvert GUI & MSConvert CMD but receiving different results in some scans (Difference1.png & Difference2.png) below are the details

**Selecting parameters using MSConvert GUI**

[image]

**Running commands in MSConvert CMD**
_with compressions_
[1] msconvert mydfoldername.d --mzXML --v --zlib
[2] msconvert mydfoldername.d --mzXML --v --64 --zlib
[3] msconvert mydfoldername.d --mzXML --v --64 --mz64 --inten64 --zlib

_without compression_
[1] msconvert mydfoldername.d --mzXML --v

Below are the highlighted results (  left side results _MSConvert CMD_ & right side results _MSConvert GUI_ ) obtained after running first command _with compressions_

Result 1/2
[image]

Result 2/2
[image]

If you need any other information then please feel free to ask.
Thank you very much.

chambm, 2022-12-05:
What does the version information at the top of the converted files say? I suspect somehow the versions being run are not the same.

imeMFK01, 2022-12-05:
Dear Chambm,

As per my understanding based on your question please find below the information also both files are showing same (below) data.

**<?xml version="1.0" encoding="ISO-8859-1"?>
<mzXML xmlns="http://sashimi.sourceforge.net/schema_revision/mzXML_3.2"
       xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
       xsi:schemaLocation="http://sashimi.sourceforge.net/schema_revision/mzXML_3.2 http://sashimi.sourceforge.net/schema_revision/mzXML_3.2/mzXML_idx_3.2.xsd">**
       

       **<dataProcessing>
      <software type="conversion" name="ProteoWizard software" version="3.0.22298"/>**

Thanks.

chambm, 2022-12-05:
Hmm. I'm not really sure why they differ then but it doesn't look like the difference is anything more than floating point precision issues. Are you running MSConvertGUI on the mzXML from msconvert.exe, or running both on the .d?

imeMFK01, 2022-12-05:
### ...more than floating point precision issues... 
Yes, it is not only based on the **floating point precision issues** but also both files' having different **compressedLen** 

_MSCONVERT CMD_
`lowMz="322.048121001651"`
`highMz="1357.633846019402"`
`compressedLen="10055"`

_MSCONVERT GUI_
`lowMz="322.048121"`
`highMz="1357.63384601654"`
`compressedLen="10052"`

### ...Are you running MSConvertGUI on the mzXML from msconvert.exe, or running both on the .d?...
I am using MSConvertGUI for converting .d folder to mzXML and same doing (.d folder to mzXML) for the msconvert.exe by using cmd.

Thanks.

chambm, 2022-12-05:
If there is some floating point precision issue then it could affect the binary numbers that are being encoded into base64. The binary is always base64 encoded: the zlib compression is optional.

imeMFK01, 2022-12-05:
Yes, there is an issue of **floating point precision** and also **compressedLen** as well. So, can you please let me know how its possible to resolve this?

**[Recursion]**
 - Parameters and input file are same 
 - Most of the results (to be more precisely ~95% results are same)
 
So, how it is possible that I am getting 5% different results? Moreover, can you please let me know if there are some default parameters which maybe already hard coded in _MSConvert GUI_ & _MSConvert CMD (msconvert.exe)_?

Really thanks for your valuable time.