ProteoWizard/pwiz #2408: Using the scan filter in proteowizard for thermo files, the mzML files contains no scans

michelleemoss, 2022-12-05:
Used the MSConvert GUI version 3.0.22336 to generate an mzML of thermo file with the scan filter on. Then opened new mzML file into SeeMS and it contains no scans. 

[image]

[image]

chambm, 2022-12-05:
Try the SIM as Spectra option.

michelleemoss, 2022-12-05:
@chambm Just tried that and I'm seeing the same results

michelleemoss, 2022-12-06:
@chambm Matt, here is a link to the thermo raw file in question. Thanks 
https://drive.google.com/drive/folders/1Vl9rybBRgieySclGiBgM4Piqy90A6BsD

chambm, 2022-12-06:
What version of pwiz? I'm able to filter that file on scan numbers 5-10 with no issue.

michelleemoss, 2022-12-06:
@chambm version 3.0.22336

PR #2411 (2022-12-06): - fixed scanNumber filter not working for Thermo
Reported by Michelle.
Closes #2408
closed 2022-12-06 by chambm