ProteoWizard/pwiz #2434: CWT centroiding

christophuv, 2022-12-15:
Hello

Thank you for the great MSConvert processing tool! 

I would like to convert some data acquired in the profile mode data to the centroid mode data and the mzML format. As the vendor algorithm for centroiding is not supported, the only other option is the CWT filter for peakPicking. However, I am unable to do that correctly. 
The conversion itself works with this filter (only filter: peakPicking filter with CWT, see screenshot below) and in the centroided data there is only a single mz value for each of the profile mode mz distributions. However, these mz values are exactly those of the most abundant mz signal in the profile mode data and no interpolation is carried out. 

[image]
The grey dots show the profile mode continuous distribution (not colored, just to show the discrete mz values recorded in this mode) and the colored dots show the selected mz values after centroiding. It can be seen that no interpolation is carried out. 

I tried to understand the respective code and I think I found it (if/else in https://github.com/ProteoWizard/pwiz/blob/0fdbda9236fd25181464c305549cf75a0ec4fd03/pwiz/analysis/common/CwtPeakDetector.cpp starting on line 483). If centroid_ is set to true, this interpolation should be calculated, however, centroid_ seems to be set to false per default and I don't know how to change that. 

My parameters for conversion and centroiding are: 

[image]

Am I doing anything wrong? 

Thank you for your help

Best regards, christoph

chambm, 2022-12-15:
Try the "Show Command Line" option and then copy it to a command-line window, then add the centroid=true parameter inside `--filter "peakPicking cwt snr=0.1 peakSpace=0.1 msLevel=1-"`
e.g.
`--filter "peakPicking cwt snr=0.1 peakSpace=0.1 msLevel=1- centroid=true"`
The GUI doesn't have that option currently. I'd be interested if that fixes the issue. It probably will, because IIUC the weighted average it's doing is very similar to doing interpolation.

christophuv, 2022-12-16:
@chambm: thank you very much for you quick answer

That solved the issue. Thanks a lot! 

Have a good day!

closed 2022-12-16 by chambm