ProteoWizard/pwiz #2460: subset feature fails to separate the assigned "Scan number"

sajfb, 2023-01-18:
I have *.D* data files that can be easily converted into *.mzML* format; however, the subset feature cannot generate *.mzML* files only from the assigned scan numbers or scan time boundaries.

sajfb, 2023-01-18:
Here is a link to the data (*.D* format) from metabolomics workbench:

https://www.metabolomicsworkbench.org/data/DRCCStudySummary.php?Mode=SetupRawDataDownload&StudyID=ST001190

sajfb, 2023-01-18:
I tried using the **Scan time** boundaries to isolate the desired scans and it worked, but the *Scan number* can not work to isolate the desired scans.

closed 2023-01-18 by sajfb

chambm, 2023-01-18:
Glad you got it working. Agilent does have scan numbers but they are not contiguous. Look at the .D or mzML file in SeeMS to see what they are.