ProteoWizard/pwiz #2463: Make an ARM image available

Adafede, 2023-01-19:
Hi,

I looked at https://hub.docker.com/r/chambm/pwiz-skyline-i-agree-to-the-vendor-licenses, which is really great but there is no ARM image available. Would it be feasible to push one?

Best,

chambm, 2023-01-19:
No, because Wine doesn't work on ARM and even if it did, the vendor libraries we use to read vendor data are x86 as well. If you know of some emulation layer we can put in so that Wine thinks it's running on x86, that's a possibility.

closed 2023-01-19 by chambm

Adafede, 2023-06-01:
Don't know if suits your needs, but https://github.com/AndreRH/hangover or https://github.com/ptitSeb/box86 eventually?

chambm, 2023-06-06:
Does Hangover work for 64-bit Wine?

ptitSeb, 2023-06-06:
Box86 and Box64 are compatible with x86/x86_64 AppImage (it that helps)

chambm, 2023-06-09:
Box64 does look like a possibility. But I don't have any ARM hardware to test this with so someone who does would have to take the lead.

Adafede, 2023-06-09:
I can test, but don't have the necessary background to implement it

Adafede, 2023-09-18:
I did not manage doing it, but here is what I started with...maybe it can help:

install.sh:
```bash
#!/bin/bash

	# NOTE: Can only run on aarch64 (since box64 can only run on aarch64)
	# box64 runs wine-amd64, box86 runs wine-i386.

	### User-defined Wine version variables ################
	# - Replace the variables below with your system's info.
	# - Note that we need the amd64 version for Box64 even though we're installing it on our ARM processor.
	# - Note that we need the i386 version for Box86 even though we're installing it on our ARM processor.
	# - Wine download links from WineHQ: https://dl.winehq.org/wine-builds/
  
	local branch="devel" #example: devel, staging, or stable (wine-staging 4.5+ requires libfaudio0:i386)
	local version="7.1" #example: "7.1"
	local id="debian" #example: debian, ubuntu
	local dist="bullseye" #example (for debian): bullseye, buster, jessie, wheezy, ${VERSION_CODENAME}, etc 
	local tag="-1" #example: -1 (some wine .deb files have -1 tag on the end and some don't)

  ########################################################

	# Clean up any old wine instances
	wineserver -k # stop any old wine installations from running
	rm -rf ~/.cache/wine # remove any old wine-mono/wine-gecko install files
	rm -rf ~/.local/share/applications/wine # remove any old program shortcuts

	# Backup any old wine installs
	rm -rf ~/wine-old 2>/dev/null; mv ~/wine ~/wine-old 2>/dev/null
	rm -rf ~/.wine-old 2>/dev/null; mv ~/.wine ~/.wine-old 2>/dev/null
	sudo mv /usr/local/bin/wine /usr/local/bin/wine-old 2>/dev/null
	sudo mv /usr/local/bin/wine64 /usr/local/bin/wine-old 2>/dev/null
	sudo mv /usr/local/bin/wineboot /usr/local/bin/wineboot-old 2>/dev/null
	sudo mv /usr/local/bin/winecfg /usr/local/bin/winecfg-old 2>/dev/null
	sudo mv /usr/local/bin/wineserver /usr/local/bin/wineserver-old 2>/dev/null

	# Wine download links from WineHQ: https://dl.winehq.org/wine-builds/
	LNKA="https://dl.winehq.org/wine-builds/${id}/dists/${dist}/main/binary-amd64/" #amd64-wine links
	DEB_A1="wine-${branch}-amd64_${version}~${dist}${tag}_amd64.deb" #wine64 main bin
	DEB_A2="wine-${branch}_${version}~${dist}${tag}_amd64.deb" #wine64 support files (required for wine64 / can work alongside wine_i386 main bin)
		#DEB_A3="winehq-${branch}_${version}~${dist}${tag}_amd64.deb" #shortcuts & docs
	LNKB="https://dl.winehq.org/wine-builds/${id}/dists/${dist}/main/binary-i386/" #i386-wine links
	DEB_B1="wine-${branch}-i386_${version}~${dist}${tag}_i386.deb" #wine_i386 main bin
	DEB_B2="wine-${branch}_${version}~${dist}${tag}_i386.deb" #wine_i386 support files (required for wine_i386 if no wine64 / CONFLICTS WITH wine64 support files)
		#DEB_B3="winehq-${branch}_${version}~${dist}${tag}_i386.deb" #shortcuts & docs

	# Install amd64-wine (64-bit) alongside i386-wine (32-bit)
	echo -e "Downloading wine . . ."
	wget -q ${LNKA}${DEB_A1} 
	wget -q ${LNKA}${DEB_A2} 
	wget -q ${LNKB}${DEB_B1} 
	echo -e "Extracting wine . . ."
	dpkg-deb -x ${DEB_A1} wine-installer
	dpkg-deb -x ${DEB_A2} wine-installer
	dpkg-deb -x ${DEB_B1} wine-installer
	echo -e "Installing wine . . ."
	mv wine-installer/opt/wine* ~/wine
	
	# Download wine dependencies
	# - these packages are needed for running box86/wine-i386 on a 64-bit RPiOS via multiarch
	sudo dpkg --add-architecture armhf && sudo apt-get update # enable multi-arch
	sudo apt-get install -y libasound2:armhf libc6:armhf libglib2.0-0:armhf libgphoto2-6:armhf libgphoto2-port12:armhf \
		libgstreamer-plugins-base1.0-0:armhf libgstreamer1.0-0:armhf libldap-2.4-2:armhf libopenal1:armhf libpcap0.8:armhf \
		libpulse0:armhf libsane1:armhf libudev1:armhf libusb-1.0-0:armhf libvkd3d1:armhf libx11-6:armhf libxext6:armhf \
		libasound2-plugins:armhf ocl-icd-libopencl1:armhf libncurses6:armhf libncurses5:armhf libcap2-bin:armhf libcups2:armhf \
		libdbus-1-3:armhf libfontconfig1:armhf libfreetype6:armhf libglu1-mesa:armhf libglu1:armhf libgnutls30:armhf \
		libgssapi-krb5-2:armhf libkrb5-3:armhf libodbc1:armhf libosmesa6:armhf libsdl2-2.0-0:armhf libv4l-0:armhf \
		libxcomposite1:armhf libxcursor1:armhf libxfixes3:armhf libxi6:armhf libxinerama1:armhf libxrandr2:armhf \
		libxrender1:armhf libxxf86vm1 libc6:armhf libcap2-bin:armhf # to run wine-i386 through box86:armhf on aarch64
[... 84 lines cut ...]
RUN winecfg

# Install .NET, Windows components, fonts, etc.
RUN winetricks -q dotnet48 && wineserver -w && winetricks -q win7 && xvfb-run winetricks -q vcrun2008 vcrun2017 corefonts && wineserver -w && rm -fr /root/.cache/winetricks

# Could not test it?
# COPY dir:b59a1bb699b1fb537248f5d825117750c24a83f2bc35bb5cca983a806b2a8c54 in /wineprefix64/drive_c/pwiz 

# Set environment variables for Wine
ENV WINEDEBUG=-all
ENV WINEPATH=C:\pwiz;C:\pwiz\skyline

# Additional setup if required
RUN apt-get update && apt-get -y install sudo && apt-get -y clean && echo "ALL     ALL=NOPASSWD:  ALL" >> /etc/sudoers && echo '#!/bin/sh\nsudo -E -H -u root wine64 "$@"' > /usr/bin/wine64_anyuser && echo '#!/bin/sh\nsudo -E -H -u root wine "$@"' > /usr/bin/wine_anyuser && chmod ugo+rx /usr/bin/wine*anyuser && rm -rf /var/lib/apt/lists/* /usr/share/doc /usr/share/doc-base /usr/share/man /usr/share/locale /usr/share/zoneinfo

# Create a directory for data
RUN mkdir /data

# Set the working directory
WORKDIR /data

# Specify the default command (you can change this to your application)
CMD ["wine64_anyuser", "msconvert"]

```

Adafede, 2023-09-20:
Oh, I think I found the needed gem:

https://gitlab.com/Linaro/windowsonarm/woa-linux/-/blob/master/containers/unified.Dockerfile