ProteoWizard/pwiz #2489: Coversion of casanovo mztab to blib format

kostrouc, 2023-02-08:
I am attempting to convert mztab peptide identification from casanovo to .blib format but have been unsuccessful using BlibBuild. The first error I recieved was:

'''
Reading results from casnovo_output.mztab.txt
ERROR: Failed to open input file '/home/kostrouchov/unimod.xml'.
100%
'''
I then downloaded the unimod.xml file from: https://www.unimod.org/downloads.html

After running the script again I received the error:

```
(base) kostrouchov@myip:~$ ./BlibBuild casnovo_output.mztab.txt casanovo_convert.blib
Reading results from casanovo_output..mztab.txt
ERROR: No acceptable score type found
ERROR: reading file casanovo_output.mztab.txt
100%
```

I have attached the example .mzTab file I am working with. Any help would be appreciated. Thank you!

[casanovo_output.mztab.txt](https://github.com/ProteoWizard/pwiz/files/10689695/casanovo_output.mztab.txt)

nickshulman, 2023-02-08:
There are a couple of things wrong with that mztab.txt file:
The line which is supposed to have a score looks like this:
MTD	psm_search_engine_score[1]	"[MS, MS:1001143, search engine specific score for PSMs, ]"
The only score types which BlibBuild.exe accepts are:
MS:1001491, percolator:Q value
MS:1001330, X!Tandem:expect
MS:1001172, Mascot:expectation value
MS:1001901, MaxQuant:PEP
MS:1002054, MS-GF:QValue
MS:1002354, PSM-level q-value

The other problem is that there are tab characters at the end of each line. 
BlibBuild.exe cannot handle extra tabs at the end of the line with the score type. You should remove all of the extra tabs at the end of lines.

(I don't actually know anything mztab files or what they are supposed to look like, but I hope this helps)

kostrouc, 2023-02-08:
Can the score-type be specified as UNKNOWN like in the .ssl files?

nickshulman, 2023-02-09:
I don't think there is a way to tell BiblioSpec to ignore the scoring, but you could lie about what the scores are.

Here is a file where I changed the search engine score line so that it says it used "PSM-level q-value" and I replaced all the actual scores with the number 0.
[fixed.mztab.txt](https://github.com/ProteoWizard/pwiz/files/10691396/fixed.mztab.txt)
I was not able to actually test this because I did not have the file "mymgffile.mgf".

kostrouc, 2023-02-09:
I attempted this again with an mgf_example.mgf file and a new100.mztab.txt file, and it seems that nothing is passing the score or they are all ambiguous which does not make sense:

```
(base) kostrouchov@ip:~$ ./BlibBuild new100.mzTab out.blib
Reading results from new100.mzTab.
Using score 'MS:1002354, PSM-level q-value'
100%
No matches left after removing ambiguous spectra in /home/kostrouchov/mgf_example.mgf.
Loading /home/kostrouchov/mgf_example.mgf.
Reading spectra from /home/kostrouchov/mgf_example.mgf.
ERROR: No spectra were found for the new library.
```

I see there is a -K option to ignore ambiguity. Using this option resulted in a similar error:

```
(base) kostrouchov@ip:~$ ./BlibBuild -K new100.mzTab out.blib
Reading results from new100.mzTab.
Using score 'MS:1002354, PSM-level q-value'
100%
Loading /home/kostrouchov/mgf_example.mgf.
Reading spectra from /home/kostrouchov/mgf_example.mgf.
ERROR: No spectra were found for the new library.
```

What else would you recommend I change for this to be successful in creating a ./blib file?

I added .txt extension to the mgf so it could be attached. unimod.xml can be downloaded here: https://www.unimod.org/downloads.html

[new100.mztab.txt](https://github.com/ProteoWizard/pwiz/files/10700512/new100.mztab.txt)
[mgf_example.mgf.txt](https://github.com/ProteoWizard/pwiz/files/10700515/mgf_example.mgf.txt)

kostrouc, 2023-02-13:
Does the .ssl format utilize the precursor m/z information? I didn't see this incorporated into the example .ssl datasets. I notice when I place the "scan", "charge", "sequence", "score", "file", and "score-type" into a .ssl file it generates a .blib file but I can't confirm if it ever utilized that information.

nickshulman, 2023-02-14:
> Does the .ssl format utilize the precursor m/z information? I didn't see this incorporated into the example .ssl datasets. I notice when I place the "scan", "charge", "sequence", "score", "file", [...]

No, the precursor m/z is not part of the ssl format.
Skyline does not care about the precursor m/z number that can be found in a .blib file. 
Skyline always calculates the theoretical m/z of the peptide based on the peptide sequence and charge.