ProteoWizard/pwiz #2511: Interference in Thermo data from Easy-IC mass calibration !!

Sanchezillana, 2023-02-21:
Hi! 

I have just found an interesting issue in our data from Thermo Orbitrap Exploris and Thermo Orbitrap Tribrid when using Easy-IC internal mass calibration. When I opened the .raw files with Thermo software (FreeSync) the ion from fluoranthene [m/z=202.0788] (the calibrant reagent) is removed from the MS1 and MS2 data, but msconvert does not removes that ion. Is there a way to remove that mass with the current implemented filters? As fas as I know is the only ion coming from Easy-IC internal mass calibration but if there are more coming from that calibrant would be worse...

Check one of my blanks to see the problem: 
https://universitatdevalencia-my.sharepoint.com/:i:/g/personal/angel_illana_uv_es/EQE16VML5hNNmRlpC9P10-EBn7p4G39DMywG6-z_mp-YNw?e=BxrprH

Ángel SI

chambm, 2023-02-21:
This is probably due to exception peaks being enabled. Another user had their peak of interest removed for being too close to the lockmass peak. It's much worse to need the peak and not have it than to have the peak and not need it, right? When you have some unwanted peak that you know the m/z of, it is theoretically trivial to exclude it with a filter. Unfortunately the only filter we have now that does something related to that doesn't have an exclude mode: it can only include a given m/z window, not exclude it (i.e. it excludes everything outside the window). When I get a chance I will address that but I can't promise it'll be in the immediate future.

Sanchezillana, 2023-02-21:
> This is probably due to exception peaks being enabled. Another user had their peak of interest removed for being too close to the lockmass peak. It's much worse to need the peak and not have it than [...]

Yeah, ofc! It is always worse losing information. Thank yoy @chambm for taking this into account! 
This peak I think could be critical for some MSMS matching metrics such as the cosine score in molecular networking, for example. I am quite surprised when I found that the exclusion of certain masses is not a common filter avaliable in msconvert neither in other programs such as MzMine. Easy for me to clean a certain mass from a mgf file but not sure if I will manage to clean it from a mzml...

Sanchezillana, 2023-02-21:
I've cheked and yes, with this "Exception peaks" option enabled on FreeSync I can see the calibrant peaks. Apart of 202.0788 also other at 173.4681. Does anyone know where I can find the full list of this Easy-IC calibrant peaks? This "exception peak" information is encoded on the mzml files or just in the .raw? 

[image]

chambm, 2023-02-21:
Just in the RAW.

lfnothias, 2023-02-22:
Thanks @Sanchezillana
We have just observed the very same phenomenon/issue with Exploris 120 data and described here:
https://github.com/compomics/ThermoRawFileParser/issues/153
I have an issue with file conversion from raw to mzML with MSConvert but I presume this is a different problem (since you could convert your file correctly).

Note that, as reported here https://github.com/compomics/ThermoRawFileParser/issues/153, conversion to mzML with ThermoRawFileParser similarly doesn't remove the lockmass, however conversion with RawConverter seems to remove correctly the lockmass (for mzXML file only).