ProteoWizard/pwiz #2512: mzML conversion fails with Exploris data

lfnothias, 2023-02-22:
Hi pwiz developers/community,

First of all, thanks a lot for the awesome software and support !

Basically trying to convert the .raw file from Exploris 120 with MSConvert produces this error:

```
Starting...
Opening file "J:\20230217_RMG_RMG_1_conditioning_26.raw" for read...
Calculating SHA1 checksum...
Processing...
Failed - System.Runtime.InteropServices.SEHException (0x80004005): External component has thrown an exception.
   at pwiz.analysis.SpectrumListFactory.wrap(MSData* , vector<std::basic_string<char\,std::char_traits<char>\,std::allocator<char> >\,std::allocator<std::basic_string<char\,std::char_traits<char>\,std::allocator<char> > > >* , IterationListenerRegistry* )
   at pwiz.CLI.analysis.SpectrumListFactory.wrap(MSData msd, IList`1 wrappers, IterationListenerRegistry ilr)
   at MSConvertGUI.MainLogic.processFile(String filename, Config config, ReaderList readers, Map`2 usedOutputFilenames)
   at MSConvertGUI.MainLogic.Go(Config config, Map`2 usedOutputFilenames)
```

The command used (Version: 3.0.22356-c1a5b0f (automated build)) is:
```
"C:\Users\AppData\Local\Apps\ProteoWizard 3.0.22356.c1a5b0f 64-bit\msconvert.exe" --32 --filter "peakPicking vendor msLevel=1-2" --filter "msLevel 1-2" --filter "titleMaker <RunId>.<ScanNumber>.<ScanNumber>.<ChargeState> File:"""^<SourcePath^>""", NativeID:"""^<Id^>""""  "J:\20230217_RMG_RMG_1_conditioning_26.raw"
```

NB0: we do have other channels for UV/CAD in the raw file.

Other NBs relevant for the EASY-IC issue just reported by @Sanchezillana https://github.com/ProteoWizard/pwiz/issues/2511

**NB1: Converting with ThermoRawFileParser works fine but the lockmass is still present in MS1/MS2.** 
**NB2: Converting with with RawConverter produces an mzXML file with the lockmass removed (trying mzML format doesn't create an output file)**. [(http://fields.scripps.edu/rawconv/](http://fields.scripps.edu/rawconv/)

The files can be accessed here  [https://drive.switch.ch/index.php/s/y35gSbwxpn8sk8s]( https://drive.switch.ch/index.php/s/y35gSbwxpn8sk8s)

Any idea what could cause that ?

chambm, 2023-02-22:
It crashes on PDA scan 6002 and when trying to get the PDA chromatogram. When I view that scan in QualBrowser it's blank, and when I try to view the scan header it crashes QualBrowser. So, definitely some file corruption right at the end. Luckily you can work around it with filters to exclude scan 6002 and the PDA chromatogram: `--filter "peakPicking vendor msLevel=1-2" --filter "scanNumber 0-6001" --filter "msLevel 1-2" --chromatogramFilter "index 0-2"` worked for me.

closed 2023-02-22 by chambm

lfnothias, 2023-02-22:
That works perfectly ! Thanks a lot for your help !