ProteoWizard/pwiz #2527: msconvert returned an error when access for some files

FloydKu, 2023-03-10:
I'm running msconvert with docker images.
For some of my .raw files , I get the following message.

processing file: /data\HF180425_RN_N_Lu10.raw
Error opening terminal: unknown.

```
Unhandled Exception: System.AccessViolationException: Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
   at System.Threading.SemaphoreSlim.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at System.Collections.Concurrent.BlockingCollection`1.TryTakeWithNoTimeValidation(T& item, Int32 millisecondsTimeout, CancellationToken cancellationToken, CancellationTokenSource combinedTokenSource)
   at System.Collections.Concurrent.BlockingCollection`1.<GetConsumingEnumerable>d__68.MoveNext()
   at ThermoFisher.CommonCore.RawFileReader.StructWrappers.Scan.TrailerScanEvents.<>c__DisplayClass25_2.<Load>b__1()
   at System.Threading.Tasks.Task.Execute()
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot)
   at System.Threading.Tasks.Task.ExecuteEntry(Boolean bPreventDoubleExecution)
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
wine: Unhandled page fault on execute access to 0x00000000 at address (nil) (thread 0043), starting debugger...
Can't attach process 0038: error 5
0049:err:winediag:nodrv_CreateWindow Application tried to create a window, but no driver could be loaded.
0049:err:winediag:nodrv_CreateWindow Make sure that your X server is running and that $DISPLAY is set correctly.
```

chambm, 2023-03-13:
Do you also get an error when converting those files on a Windows machine?

FloydKu, 2023-03-14:
> Do you also get an error when converting those files on a Windows machine?

I have retride the same operation with those files，and the error does not appear again.

chambm, 2023-03-14:
Yeah wine can be flaky sometimes especially with Thermo RAWs.

closed 2023-03-14 by chambm