ProteoWizard/pwiz #2537: Run ID contains a single unicode character

bernt-matthias, 2023-03-20:
When I convert a single raw file (that starts with a number) I sometimes observe that the first character appears unicode encoded in the mzML file, e.g.: 

```
<run id="_x0031_90615_NJ_Taubert_13CT1C_FS_30sec" ...
```

which crates problems in downstream analyses. If I use a non numeric character as the 1st character of the file name the problem does not occur.

I'm using the latest msconvert from usegalaxy.eu .. can share the history if needed.

chambm, 2023-03-20:
That's because run::id is a xs:ID. It's not allowed to start with anything except [A-Za-z] and underscore. But actually mzML just released 1.1.1 which changes it to a xs:string. I haven't adjusted pwiz yet to take advantage of that but you can either decode the id in the file reader (replace _x0031_ with char val 31 which is 1 I think) or wait for an update (not sure when I'll get around to updating the Galaxy version though). Or maybe just have galaxy prepend _ to the filename before passing it to msconvert. Beware though that the middle of the xs:ID can only have [A-Za-z0-9_-], so not much more permissive of symbols and stuff.

bernt-matthias, 2023-03-20:
For now it suffices to know why this happens. 

Let me know if I can help with a new Galaxy version :)