ProteoWizard/pwiz #2551: imzML ontology in Proteowizard

kpnovoselov, 2023-04-04:
Is it possible to inlcude imzml ontology (https://ms-imaging.org/imzml/ or https://github.com/imzML/imzML) as a part of ProteoWizard? It is almost the same as mzml fromat with some MS imaging related information and binary data stored in the separate file.

I am using ProteoWizard for reading vendor files and for writing imzml files with extended writer derived from  pwiz::minimxml::XMLWriter, but I need to re-generate ontology related source files and recompile ProteoWizard for each new version of pwiz.

chambm, 2023-04-04:
Do I understand correctly you want https://github.com/imzML/imzML/blob/master/imagingMS.obo to be included in cv.hpp/cpp?

kpnovoselov, 2023-04-04:
Yes