ProteoWizard/pwiz #2621: boost / create dir issue (version 3.0.23144)

apulvino, 2023-06-15:
boost::filesystem::create_directories: Access denied: "/path/to/dir/above/data_files"
Please report this error to support@proteowizard.org.
Attach the command output and this version information in your report:

ProteoWizard release: 3.0.23144 (0ae01f1)
Build date: May 24 2023 21:31:43

FULL CMD:
singularity exec --bind $PWD/raw_files:$HOME/Downloads/files \
--env WINEPREFIX=$HOME/Downloads/.wine \
$HOME/Downloads/pwiz-skyline-i-agree-to-the-vendor-licenses_latest.sif \
wine /wineprefix64/drive_c/pwiz/msconvert.exe $HOME//Downloads/files/* -o $PWD/test_dir

chambm, 2023-06-15:
Try running with the singularity option to make the image read-write.

apulvino, 2023-06-15:
FATAL: no SIF writable overlay partition found

Still not letting me have it unfortunately

apulvino, 2023-06-16:
was able to resolve a solution re: https://github.com/jspaezp/elfragmentador-data#setting-up-msconvert-on-singularity-

closed 2023-06-16 by apulvino