ProteoWizard/pwiz #2623: "error in thread: Invalid Scan Number" for IMS

PierreSnell, 2023-06-19:
Hello, 

First thanks a lot and congrats for this tool. 

I'm trying to convert Ion mobility spectrometry (IMS) data but got a : 
```
2023-06-19T13:55:14.627609472Z [SpectrumWorkerThreads::work] error in thread: Invalid Scan Number
```

I'm converting with the latest docker and : 

```
wine msconvert my_super_IMS_file.raw --outdir /out_data --mzML --zlib --numpressAll --32
```

I might also be using the wrong parameter for this task...

Thanks in advance for any help. 
Have a great day.

chambm, 2023-06-21:
Hmm, weird error. I'd want to try it on Windows native machine and if it happens there, debug it. Why the scanNumber filter would solve it I don't know except maybe you've filtered out one or more problematic scans with your filter.

PierreSnell, 2023-06-21:
Hello @chambm, 

Unfortunately, this was some Python code calling the Docker API (to have some automation) all on a Linux machine.
I might try it natively on a Windows machine... 

It seems weird as the resulting mzml data is not convincing to me... (it seems only that some of the data is here).  

So if I get you right, I can just run `wine msconvert my_super_IMS_file.raw --outdir /out_data` on Ionic Mobility raw data and everything should work out of the box? 

I will investigate fully all my raw data to be sure that there is no corrupted scans. 

Thanks for the fast reply !

chambm, 2023-06-21:
Yes it should work with no filters. Generally the combinedIonMobilitySpectra mode is recommended but both should work.

PierreSnell, 2023-06-27:
Hello, I found time to try on a windows machine. 

Downloaded the latest `proteoWizard`: https://proteowizard.sourceforge.io/download.html 
Ran the GUI, on the same files. 

Here is the output : 

```
D:\path\to\input\MySuperFile.raw
------------------------------
Starting...
Opening file "D:\path\to\input\MySuperFile.raw" for read...
Calculating SHA1 checksum...
Processing...
Writing "D:\path\to\output\MySuperFile.mzML"...
Skipping spectrum 0 "merged=1 function=3 block=1": Invalid Scan Number
Skipping spectrum 93 "merged=94 function=3 block=2": Invalid Scan Number
Skipping spectrum 186 "merged=187 function=3 block=3": Invalid Scan Number
Skipping spectrum 279 "merged=280 function=3 block=4": Invalid Scan Number
Skipping spectrum 372 "merged=373 function=3 block=5": Invalid Scan Number
Skipping spectrum 465 "merged=466 function=3 block=6": Invalid Scan Number
Skipping spectrum 558 "merged=559 function=3 block=7": Invalid Scan Number
Skipping spectrum 651 "merged=652 function=3 block=8": Invalid Scan Number
Skipping spectrum 744 "merged=745 function=3 block=9": Invalid Scan Number
Skipping spectrum 837 "merged=838 function=3 block=10": Invalid Scan Number
Skipping spectrum 930 "merged=931 function=3 block=11": Invalid Scan Number
Skipping spectrum 1023 "merged=1024 function=3 block=12": Invalid Scan Number
Skipping spectrum 1116 "merged=1117 function=3 block=13": Invalid Scan Number
Skipping spectrum 1209 "merged=1210 function=3 block=14": Invalid Scan Number
Skipping spectrum 1302 "merged=1303 function=3 block=15": Invalid Scan Number
Skipping spectrum 1395 "merged=1396 function=3 block=16": Invalid Scan Number
Skipping spectrum 1488 "merged=1489 function=3 block=17": Invalid Scan Number
Skipping spectrum 1581 "merged=1582 function=3 block=18": Invalid Scan Number
Skipping spectrum 1674 "merged=1675 function=3 block=19": Invalid Scan Number
Skipping spectrum 1767 "merged=1768 function=3 block=20": Invalid Scan Number
Skipping spectrum 1860 "merged=1861 function=3 block=21": Invalid Scan Number
Skipping spectrum 1953 "merged=1954 function=3 block=22": Invalid Scan Number
Skipping spectrum 2046 "merged=2047 function=3 block=23": Invalid Scan Number
Skipping spectrum 2139 "merged=2140 function=3 block=24": Invalid Scan Number
Skipping spectrum 2232 "merged=2233 function=3 block=25": Invalid Scan Number
Skipping spectrum 2325 "merged=2326 function=3 block=26": Invalid Scan Number
Skipping spectrum 2418 "merged=2419 function=3 block=27": Invalid Scan Number
Skipping spectrum 2511 "merged=2512 function=3 block=28": Invalid Scan Number
Finished.
```

Here is the command line : 

```
"C:\Program Files\ProteoWizard\ProteoWizard 3.0.23178.fa38203\msconvert.exe" --32 --zlib --numpressLinear --combineIonMobilitySpectra --filter "titleMaker <RunId>.<ScanNumber>.<ScanNumber>.<ChargeState> File:"""^<SourcePath^>""", NativeID:"""^<Id^>""""  "D:\path\to\input\MySuperFile.raw"
```

So, with "exactly" the same arguments, the Linux docker one is failing... any idea on how to fix it? 
I can do a PR if needed, but I would need the docker for automation as the GUI is not scalable for me. 

Thanks in advance,

chambm, 2023-06-27:
That looks like maybe the lockmass function isn't being handled properly. Maybe msconvert is trying to treat it like it has ion mobility when it doesn't. Can you share your RAW file?

PierreSnell, 2023-06-27:
I sent you a WeTransfer (because of the size) to your email address (the one on your GitHub).

chambm, 2023-06-28:
It was the opposite: the lockmass function was being treated like it didn't have ion mobility when it actually does. File converts fine with that fixed in the linked PR which I'll merge if nothing comes up in review.

closed 2023-06-30 by chambm

PierreSnell, 2023-07-04:
Hello, 
I was wondering, as the PR is merged (thanks for the quick fix !) if you could update the docker image :) 

Thanks in advance, 
Have a nice day.

PR #2636 (2023-06-30): * fixed errors caused by assuming lockmass function is not IMS
* fixed errors caused by assuming lockmass function is not IMS: Invalid Scan Number exceptions (reported by Pierre)

Brian, do you remember what was the impetus for that particular usage of the isLockMassFunction condition? Why not rely on the `IonMobilityByFunctionIndex()` metadata?

Closes #2623